serialport/node-serialport #549: Temporary Primary Maintainer Needed

voodootikigod, 2015-07-05:
All,

It is with a heavy heart that I have to ask the community for support of this module over the next couple weeks to months. Due to medical issues, I have to turn my complete and total attention to the health, stability, and well-being of my wife and family. As such, I am withdrawing my involvement in this project and others in order to best focus on them.

I need someone to take over as the primary maintainer for this project and guide it through the time period I am absent. There isn't a tremendous amount required for this, but it is still more than I can provide at this time. Please reply to this if you are willing to take over the project.

Sorry to have to do this and thank you for understanding

Chris Williams

jacobrosenthal, 2015-07-05:
Im happy to help. Godspeed Chris.

Id love to try to tag issues a bit more and spread the work amongst a couple of us, can anyone commit to helping with tagging issues or a specific platform? @JayBeavers @giseburt @reconbot @ anybody else.

voodootikigod, 2015-07-06:
Jacob you are now the primary.

Thank you

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

giseburt, 2015-07-06:
I certainly can't commit to primary (sorry), but I'll help all I can @jacobrosenthal . Let me know how.

My email is this same username at that google platform if you wish to contact me directly.

I hope it all works out well, Chris! I certainly understand needing to focus on family.

-Rob

jacobrosenthal, 2015-07-06:
@giseburt you have commits in on Windows right? Would you mind shepherding windows platform issues?

Im largely OSX myself, though I might be carrying a linux machine full time soon, so if anyone else can take either of those ill take the other.

rwaldron, 2015-07-06:
Of course, I'm always here to nit pick and do reality checks—ping me as necessary.

giseburt, 2015-07-06:
I'm actually terrible on Windows. I'm also OS X and Linux. :/

JayBeavers, 2015-07-17:
OK, I'll jump in where needed.  My prior experience was that many of the Windows issues were deeper platform issues with nodejs/gyp/node-gyp which were real time sinks to try and address, I'll contribute where I can.

jacobrosenthal, 2015-07-17:
What I need is people to commit to jumping in now, rather than when needed. Id like to tag up issues something like os/architecture, verification needed, patch welcome, tests needed. Then find people who run each arch as their primary so its not real hassle to have them run down a bug or verify a fix.

reconbot, 2016-04-06:
I've been working with @jacobrosenthal on this the past few weeks. There are plenty of things we could still use help on and we'll be putting together a roadmap soon. I plan on continuing to help maintain for the time being.

closed 2016-04-06 by reconbot