serialport/node-serialport #553: Need pull request (550) reviewed

munyirik, 2015-07-14:
I've had the pull request below open for a few days. Is there anyone that can review and merge if it's good?
https://github.com/voodootikigod/node-serialport/pull/550

jacobrosenthal, 2015-07-14:
Im in the middle of a week of travel. Might be out of it by friday.

munyirik, 2015-07-14:
thanks @jacobrosenthal

reconbot, 2015-07-25:
According to the docs you linked it looks great, I don't have a windows machine to test on. What happened to our Windows CI?

munyirik, 2015-08-13:
There is work being tracked by https://github.com/nodejs/hardware/issues/13 for a Windows test machine. @reconbot is the criteria to merge a successful run of the tests in this repo?

/cc @orangemocha

reconbot, 2015-08-13:
Yes, the tests need to be successful before we merge. However the coverage in the c++ layer is pretty low so we want to try the changes out on an actual machine as well.

closed 2016-04-04 by reconbot