serialport/node-serialport #557: node crash when no serialport available, using fibers

CypD, 2015-07-21:
Hi
Running this code, while there is no serialport device connected while listing ports.
```
wait.launchFiber(function()
{    
    var serialport = require("serialport");
    var SerialPort = serialport.SerialPort;

    var portlist=wait.forMethod(serialport,'list');    
    console.log(portlist);
});
```
This error happens:
```
/node_modules/wait.for/waitfor.js:38
                                fiber.run();   
                                      ^
Error: ENOENT, scandir '/dev/serial/by-id'
    at Error (native)
```
This is because the directory /dev/serial  doesn't exist when there is no serial ports available, so scanning /dev/serial/by-id gives an error.
When I plug the usb FTDI device, everything works fine, it lists 1 port.

serialport.list somehow works fine and returns undefined when running without a fiber.

Platform: linux, raspbian, raspberry pi2

jaumard, 2015-08-21:
Same here 

Platform: linux, raspbian, raspberry pi2

jacobrosenthal, 2015-10-13:
This should be fixed ages ago in 7f73eb0fbab99a0662f4d71934537ea6f57b129c

closed 2015-10-13 by jacobrosenthal