serialport/node-serialport #558: Failed to install on Linux 3.18.11+ with io.js v2.4.0

CubixSystem, 2015-07-26:
```
pi@raspberrypi ~ $ node -v
v2.4.0

pi@raspberrypi ~ $ npm -v
2.13.2

```

```
pi@raspberrypi ~ $ sudo npm install serialport -g
/usr/local/bin/serialportlist -> /usr/local/lib/node_modules/serialport/bin/serialportList.js
/usr/local/bin/serialportterm -> /usr/local/lib/node_modules/serialport/bin/serialportTerminal.js

> serialport@1.7.4 install /usr/local/lib/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/2.4.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/serialport/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: 404 response downloading https://nodejs.org/dist/v2.4.0/node-v2.4.0.tar.gz
gyp ERR! stack     at Request.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:251:14)
gyp ERR! stack     at emitOne (events.js:82:20)
gyp ERR! stack     at Request.emit (events.js:169:7)
gyp ERR! stack     at Request.onRequestResponse (/usr/local/lib/node_modules/npm/node_modules/request/request.js:985:10)
gyp ERR! stack     at emitOne (events.js:77:13)
gyp ERR! stack     at ClientRequest.emit (events.js:169:7)
gyp ERR! stack     at HTTPParser.parserOnIncomingClient (_http_client.js:415:21)
gyp ERR! stack     at HTTPParser.parserOnHeadersComplete (_http_common.js:88:23)
gyp ERR! stack     at TLSSocket.socketOnData (_http_client.js:305:20)
gyp ERR! stack     at emitOne (events.js:77:13)
gyp ERR! System Linux 3.18.11+
gyp ERR! command "/usr/local/bin/iojs" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/usr/local/lib/node_modules/serialport/build/serialport/v1.7.4/Release/node-v44-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/usr/local/lib/node_modules/serialport/build/serialport/v1.7.4/Release/node-v44-linux-arm"
gyp ERR! cwd /usr/local/lib/node_modules/serialport
gyp ERR! node -v v2.4.0
gyp ERR! node-gyp -v v2.0.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/iojs /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/usr/local/lib/node_modules/serialport/build/serialport/v1.7.4/Release/node-v44-linux-arm/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/serialport/build/serialport/v1.7.4/Release/node-v44-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:764:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 3.18.11+
node-pre-gyp ERR! command "/usr/local/bin/iojs" "/usr/local/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/local/lib/node_modules/serialport
node-pre-gyp ERR! node -v v2.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/iojs /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/usr/local/lib/node_modules/serialport/build/serialport/v1.7.4/Release/node-v44-linux-arm/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/serialport/build/serialport/v1.7.4/Release/node-v44-linux-arm' (1)
npm ERR! Linux 3.18.11+
npm ERR! argv "/usr/local/bin/iojs" "/usr/local/bin/npm" "install" "serialport" "-g"
npm ERR! node v2.4.0
npm ERR! npm  v2.13.2
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
```

skibz, 2015-08-05:
i'd advise against installing globally.

CubixSystem, 2015-08-12:
@skibz  Thanks for reply I will try your advise.

skibz, 2015-08-13:
if it fails without `-g` and without `sudo`, then it could be that your npm folder has the incorrect permissions. try something like `sudo chown -R $(whoami) $(npm root -g)`

jacobrosenthal, 2015-09-12:
Closing for #578

closed 2015-09-12 by jacobrosenthal