serialport/node-serialport #563: Node 0.12+ support

riyadhalnur, 2015-08-14:
Node 0.11 can be disregarded but v0.10 and v0.12 are the stable versions at the time of writing. We should make this module support 0.12

jacobrosenthal, 2015-09-14:
Can you report observed lack of support for node 12 on a specific platform?

fivdi, 2015-09-22:
@riyadhalnur did you report this issue after reading the following statement in the readme?
>We have confirmed things are fine with 0.11.10 and earlier, but not 0.11.11+.

riyadhalnur, 2015-09-22:
@fivdi i did but now that 0.12 and actually 4.x is the stable versions, there should be support for them

@jacobrosenthal i'll post the errors when i get a chance. i'm on a mba if that helps till then.

jacobrosenthal, 2015-09-30:
@riyadhalnur yes, we dont know of any 12.x support that isnt working. so if we cant get a report Im going to close this.

jacobrosenthal, 2015-10-11:
Please post if you find any 12.x issues.

closed 2015-10-11 by jacobrosenthal