serialport/node-serialport #564: Cannot list ports on Embedded Linux

mitsos1os, 2015-08-17:
Hi, I want to use node-serialport on a router that runs a custom flavor of embedded linux. My problem is that I want to use automatic recognition of serial port device connected depending on characteristics like vendor and product ID. From taking a look at the source code, the list operation of the module, uses the ubuntu way of getting the serial ports from the auto mounted folder in /dev/serial/(by-id|by-path)... 

Unfortunately this does not work in embedded  linux. Could it be that the direct output of the commands is used for listing serial devices and parsing all info by skipping the read from /dev/serial part?

jacobrosenthal, 2015-08-17:
Can you try https://github.com/voodootikigod/node-serialport/pull/561

mitsos1os, 2015-08-17:
No, this won't work for me. The /dev/serial/by-path that was added (https://github.com/voodootikigod/node-serialport/pull/561/files) is still accessing the /dev/serial folder which does not exist.

jacobrosenthal, 2015-08-17:
We're already returning an empty array for the case which that directory doesnt exist (and presumably no devices are plugged in). So in your distro/kernel combo /dev/serial will never exist? Which distro/kernel?

mitsos1os, 2015-08-17:
Yes in my distro /dev/serial will not exist. It's embedded busybox linux from netcomm. The router is netcomm ntc-140w. The devices are listed normally on /dev/ttyUSB0-3 etc... but /dev/serial is never created. Actually in Ubuntu there is a udev rule that creates the /dev/serial folder upon a serial device connection. So this won't happen in my router

closed 2015-11-22 by jacobrosenthal