serialport/node-serialport #568: Not receiving any data

untitledlt, 2015-08-25:
I am using the same code as in example:

```javascript
serialPort.on("open", function () {
  console.log('open');
  serialPort.on('data', function(data) {
    console.log('data received: ' + data);
  });
});
```

I know that data is received on serial port because I can see with `cat /dev/rfcomm0`
but callback on `serialPort.on('data', function(data) { ... })` is never triggered.

Node v0.12.3
Ubuntu 14.04

jacobrosenthal, 2015-09-14:
Im presuming you got this figured out?

untitledlt, 2015-09-14:
Yes, I did. There was an issue with some crashed process which was locking out serial port. Reboot solved it.

jacobrosenthal, 2015-09-14:
Happy to hear that. You can occasionally get some more eyes on issues by idling in the gitter, too https://gitter.im/voodootikigod/node-serialport

closed 2015-09-14 by jacobrosenthal