serialport/node-serialport #573: Problems to install on IOJS

henricavalcante, 2015-09-07:
```
[23:06:07] henricavalcante@machine:~/git/henricavalcante/node-aiopv2/node_modules/johnny-five (master)
$ npm install serialport
|
> serialport@1.7.4 install /Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../node_modules/nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../node_modules/nan/nan.h:661:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/henricavalcante/.node-gyp/3.2.0/include/node/node_buffer.h:31:40: note: candidate function not viable: no known conversion from 'uint32_t' (aka 'unsigned int') to
      'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/node_buffer.h:43:40: note: candidate function not viable: 2nd argument ('const char *') would lose const qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/node_buffer.h:28:40: note: candidate function not viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/node_buffer.h:36:40: note: candidate function not viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:676:12: error: no viable conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:210:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'const v8::Local<v8::Object> &' for 1st argument
class Local {
      ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:210:7: note: candidate constructor (the implicit move constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&' for 1st argument
class Local {
      ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:214:13: note: candidate template ignored: could not match 'Local' against 'MaybeLocal'
  V8_INLINE Local(Local<S> that)
            ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:327:13: note: candidate template ignored: could not match 'S *' against 'v8::MaybeLocal<v8::Object>'
  V8_INLINE Local(S* that)
            ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:683:26: error: no member named 'Use' in namespace 'node::Buffer'
    return node::Buffer::Use(v8::Isolate::GetCurrent(), data, size);
           ~~~~~~~~~~~~~~^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
In file included from ../node_modules/nan/nan.h:24:
In file included from /Users/henricavalcante/.node-gyp/3.2.0/include/node/node.h:42:
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:221:5: error: assigning to 'v8::Primitive *volatile' from incompatible type 'v8::Value *'
    TYPE_CHECK(T, S);
    ^~~~~~~~~~~~~~~~
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:180:37: note: expanded from macro 'TYPE_CHECK'
    *(static_cast<T* volatile*>(0)) = static_cast<S*>(0);      \
                                    ^ ~~~~~~~~~~~~~~~~~~
../node_modules/nan/nan.h:414:12: note: in instantiation of function template specialization 'v8::Local<v8::Primitive>::Local<v8::Value>' requested here
[... 58 lines cut ...]
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 14.5.0
gyp ERR! command "/usr/local/bin/iojs" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v45-darwin-x64/serialport.node" "--module_name=serialport" "--module_path=/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v45-darwin-x64"
gyp ERR! cwd /Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport
gyp ERR! node -v v3.2.0
gyp ERR! node-gyp -v v2.0.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/iojs /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v45-darwin-x64/serialport.node --module_name=serialport --module_path=/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v45-darwin-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:764:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Darwin 14.5.0
node-pre-gyp ERR! command "/usr/local/bin/iojs" "/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport
node-pre-gyp ERR! node -v v3.2.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/iojs /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v45-darwin-x64/serialport.node --module_name=serialport --module_path=/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v45-darwin-x64' (1)
npm WARN optional dep failed, continuing serialport@1.7.4
```

imyller, 2015-09-07:
node-serialport is still depending on `NAN` v1 which is not compatible with latest io.js or upcoming node.js v4 releases.

Current `node-serialport` supports io.js up to v2.5

Few people already did `NAN` v2 patching in PRs #566 and #571

rwaldron, 2015-09-08:
With the release of node.js 4.0.0 (the merger with iojs), resolving this has become an urgent matter.

jacobrosenthal, 2015-09-12:
Closing for #578

closed 2015-09-12 by jacobrosenthal