serialport/node-serialport #577: Compilation fails on Node v4.0 (Windows, OSX)

Nesciosquid, 2015-09-11:
As you might expect, trying to install the serialport module via npm doesn't seem to work for installations of Node on v4.0 on either Windows or OSX Yosemite.

We found this out after about a third of the class I teach started complaining that they couldn't run our demo code, and after two hours of frantic debugging, we realized they were the ones who waited to install Node until today.

Uninstalling Node v4.0 and Installing v0.12.7 instead (from https://nodejs.org/en/download/releases/) fixed the problem. 

FWIW, we weren't able to get the binaries to compile on any platform, following instructions on the main page and lots of the feedback given in other issues.

jacobrosenthal, 2015-09-12:
Closing for #578

closed 2015-09-12 by jacobrosenthal

Nesciosquid, 2015-09-12:
Thanks for taking a look at this!