serialport/node-serialport #578: nodejs 4.0 support

jacobrosenthal, 2015-09-12:
It sounds like we don't want to support a 1.x branch for some time? 
I dont feel great about that, especially since we have possible outstanding issues. 
Can anyone confirm or deny these?

OSX el capitan
https://github.com/voodootikigod/node-serialport/issues/552

Port listing 
Linux
https://github.com/voodootikigod/node-serialport/issues/564
https://github.com/voodootikigod/node-serialport/pull/565
https://github.com/voodootikigod/node-serialport/pull/561

windows
https://github.com/voodootikigod/node-serialport/issues/530

Others?

Then for our 2.0 release with node 4.0 support:

Our understanding is that node 4 (v8 anyway) requires gcc 4.8. Link? 

And we're thus worried about our embedded friends (pi and beaglebone black) as theyre both 4.6.x by default. Can we confirm or deny problems here?

Just from googling, Seems like Debian Jesse will ship with 4.9.x and a jessie release of rasbian will come when theyre good and ready...
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=111144

I actually think beaglebone debian is already at jessie? (confirm anyone?)

my raspberry pi 2 raspian install says
gcc version 4.6.3 (Debian 4.6.3-14+rpi1) 

A Post install script that autoinstalls and selects gcc4.8 system wide was discussed but I feel thats pretty heavy handed. It might be preferred to detect and simply give a link to instructions on how to do it yourself or select the 1.x branch of serialport if you can't

Then to get Node 4 support well need nan 2.0, seems like @rwaldrons vote is for landing #571 over #566?

I currently don't have the access or knowledge to ship builds so well need to ping Chris when the time comes.

Anything else?

imyller, 2015-09-12:
Node.js@ARM with OpenEmbedded (Yocto, Ångstrom) is ready for this.

I'm maintaining Node.js layer for OE at https://github.com/imyller/meta-nodejs
Node.js 4.0 compiles and runs perfectly on embedded platforms.

jacobrosenthal, 2015-09-12:
@imyller can you comment on what was needed and thus what might be deficient on beaglebone and pi?

imyller, 2015-09-12:
Well, I do not know the exact status of Rasbian or other pre-built distributions, but with OE everything in the distribution for the target device is compiled from source; including build system toolchains. OE is a build system - not a binary distribution.

The latest Yocto 1.8 includes gcc 4.9 and Node.js can be cross-compiled without any issues. Only thing needed are proper cross-compilation flags and configs (like meta-nodejs build scripts provides).

I've been running Node.js and io.js builds from 0.10 all the way to latest 4.0. Personally I'm running them with Beaglebone (ARM7) and x86_64.

I would say `node-serialport` need not to worry about availability of Node.js runtime itself. It is available for embedded platform if you have proper toolchain to build it or your binary distribution supports it. The real problem here is with `node-serialport` which does not support NAN v2 required for building with io.js 3+ or Node.js 4+. 

Upgrading native bindings to NAN v2 does not break builds with earlier Node.js or io.js releases so it should be harmless upgrade.

imyller, 2015-09-12:
Use case example:

Only reason current embeded hardware product we are maintaining is holding on to io.js 2.5.0 is due to fact that `node-serialport` hasn't shipped with NAN v2.

We'll upgrade to Node.js 4.0 once `node-serialport` jumps to NANv2. 

Also, we are NOT `npm install` -ing the `node-serialport` *inside* the embedded ARMv7 target system (the target does not ship with compiler toolchains). We are cross-compiling the Node.js (io.js) on x86_64 host and running cross-compiling capable `npm install` to build `node-serialport` there. Then only prebuilt binaries are shipped with the embedded devices minimal rootfs image.

ChALkeR, 2015-09-13:
Adding to the list: https://github.com/nodejs/node/issues/2798.

fivdi, 2015-09-13:
>It sounds like we don't want to support a 1.x branch for some time? 

Hmm, I don't understand what this means.

>Then for our 2.0 release with node 4.0 support:
>
>Our understanding is that node 4 (v8 anyway) requires gcc 4.8. Link?

https://github.com/nodejs/hardware/issues/16#issuecomment-138716459

>And we're thus worried about our embedded friends (pi and beaglebone black) as theyre both 4.6.x by default. Can we confirm or deny problems here?
>
>Just from googling, Seems like Debian Jesse will ship with 4.9.x and a jessie release of rasbian will come when theyre good and ready...

The current "Release" distributions for the [Raspberry Pi](https://www.raspberrypi.org/downloads/raspbian/) and the [BeagleBone Black](http://beagleboard.org/latest-images) ship with gcc/g++ 4.6.

For the Raspberry Pi this isn't a problem as Raspbian (Debian Wheezy) ships with an optional gcc/g++ 4.8 that can be installed as described [here](https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons). Note that it's possible to have both 4.6 and 4.8 installed at the same time and switch between versions as required.

For the BeagleBone Black things are a little different. It's variant of Debian Wheezy doesn't ship the optional gcc/g++ 4.8 compilers. There are however Debian Jessie "testing" distributions available [here](https://rcn-ee.com/rootfs/bb.org/testing). The testing distributions come with gcc/g++ 4.9.

>A Post install script that autoinstalls and selects gcc4.8 system wide was discussed but I feel thats pretty heavy handed. It might be preferred to detect and simply give a link to instructions on how to do it yourself or select the 1.x branch of serialport if you can't

I, personally, don't like the idea of gcc/g++ 4.8 being automatically installed. Instructions on how to do it yourself sounds like a better idea to me.

In what scenario would it be necessary to "select the 1.x branch of serialport"? I don't think this will even be possible. If Node.js 4 is being used, and gcc/g++ 4.8 isn't available, it won't be possible to select the 1.x branch of serialport.

imyller, 2015-09-13:
Why node-serialport has to worry about what compiler is available to which platform? This is about NAN. The bindings made with NAN v2 compiles (to my knowledge) on both newer and older gcc and Node.js. Same might not apply to Node.js or v8 itself, but that isn't a node-serialport issue. 

So why not just jump to NAN v2 and let users play your what node.js or compiler combination they have. 

Correct me if I'm wrong, but using NAN v2 does not break backwards compatibility; with compilers or with Node.js versions. It just offers capability to use node-serialport with Node.js 4+ for those who are otherwise able to do so.

fivdi, 2015-09-13:
>Why node-serialport has to worry about what compiler is available to which platform?

It doesn't actually have to worry about what compiler is available to which platform if it doesn't want to. I might be wrong, but this discussion is more about moving everyone forward towards Node.js 4+ without leaving too many of them behind.

 >The bindings made with NAN v2 compiles (to my knowledge) on both newer and older gcc and Node.js. Same might not apply to Node.js or v8 itself, but that isn't a node-serialport issue. 

Correct.

>So why not just jump to NAN v2 and let users play your what node.js or compiler combination they have. 

I can imagine this happening in the not too distant future.

>Correct me if I'm wrong, but using NAN v2 does not break backwards compatibility; with compilers or with Node.js versions. It just offers capability to use node-serialport with Node.js 4+ for those who are otherwise able to do so.

There's nothing to correct here :)

keyvanfatehi, 2015-09-13:
Just want to chime in to say thanks @jacobrosenthal for specifying that node-serialport compiles in iojs v2.5.0. It installs no problem in node 0.12.x but failed for me in node 4 and iojs 3.3.x. I was not sure what was going on until I saw this thread.

N4TY, 2015-09-15:
Would this also be related to issue #579 ?

keyvanfatehi, 2015-09-15:
@N4TY Yes. I just updated #579 with what I think may help you.

tigoe, 2015-09-24:
This makes me sad. I have a bunch of examples for node-serialport and I want to be able to upgrade node (and I know students new to node will start with the standard install), but until this is working with the current version, they're all useless. I don't have a windows machine (working in OSX), but I can set one up. If you have end-user tests for people to run that'd help, please post them and I'll give them a try.

rwaldron, 2015-09-24:
> This makes me sad. 

What about this makes you sad? People are actively working on confirmations to ensure a smooth transition. There is nothing sad-making about that. Pitching in is appreciated, but editorializing isn't constructive—agreed?

tigoe, 2015-09-24:
Agreed. Sorry, didn't mean to be editorializing. If there's a list of tests to be run, let me know and I'll try if I've got the resources to run them.

rwaldron, 2015-09-24:
@tigoe glad to have your help: https://github.com/voodootikigod/node-serialport/tree/master/arduinoTest

tigoe, 2015-09-24:
Looks like the examples I usually use for testing. I haven't got a lot to add as it failed at the same place for me: all working great until I upgraded from node 0.11.x to 4.1.1, when I get more or less the same compile errors as others have:

(FWIW, OSX 10.10.5 (14F27), node v4.1.1. Serial ports working fine with Arduino 1.6.5, Coolterm, screen, etc)

```
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
```
and goes on from there. Doesn't look like anything new or surprising, but if so, let me know and I'll pull the full log. Looks like it's just confirmation that it's related to NAN.  

I have some pre-compiled installations of serialport (I tend to install it locally per project rather than globally), which fail as follows:
```
$ node ListPorts.js
module.js:338
    throw err;
    ^

Error: Cannot find module '/Users/tigoe/Documents/github/NodeExamples/SerialIntro/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64/serialport.node'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:286:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/Users/tigoe/Documents/github/NodeExamples/SerialIntro/node_modules/serialport/serialport.js:14:25)
    at Module._compile (module.js:434:26)
    at Object.Module._extensions..js (module.js:452:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
```

I saw the note above that you're looking for windows testers, which I'm not, but I've got a win7 machine sitting around that's never had node on it before, so it'd be a virgin install. It's an old netbook though, maybe 4 years old. That useful at all? Can see if we've got anything around here running win8, too.

fivdi, 2015-09-25:
@tigoe it looks like you did an `npm install serialport` which isn't going to work with Node.js v4.1.1

There are currently two pull requests for Node.js 4 support, https://github.com/voodootikigod/node-serialport/pull/566 and https://github.com/voodootikigod/node-serialport/pull/571.

https://github.com/voodootikigod/node-serialport/pull/566 can be installed with `npm install "git://github.com/ghostoy/node-serialport.git#nan1to2"`

https://github.com/voodootikigod/node-serialport/pull/571 can be installed with `npm install "git://github.com/julianduque/node-serialport.git#master"`

You could try either or both of them and report your findings.

cpustar, 2015-09-26:
@fivdi 

#566 can be installed with npm install "git://github.com/ghostoy/node-serialport.git#nan1to2"
#571 can be installed with npm install "git://github.com/julianduque/node-serialport.git#master"

Neither of these work for me on Windows.

fivdi, 2015-09-26:
@cpustar can you provide more information about what doesn't work? Is it the install that doesn't work? If so, what is the output of the install process? Or are there problems at runtime?

garethadams, 2015-09-27:
Not sure if this will be of any help but I'm running Windows 10 64-bit, Node 4.1.1, Visual Studio 2015 Community and Visual Studio 2013 Community.

My initial attempt to install with the ghostoy fork failed with the following error:

error MSB8020: The builds tools for v140 (Platform Toolset = 'v140') cannot be found.

This was solved by setting the msbuild path:
`set path=C:\Program Files (x86)\MSBuild\14.0\Bin;%path%`

After this, the install went smoothly (there were a number of warnings during build, mainly to do with type casting causing truncation) and I had no problems using serialport in node afterwards.

fivdi, 2015-09-27:
@garethadams Thanks for the feedback. If I understand things correctly, you were able to use serialport to communicate with a device after the installation?

fivdi, 2015-09-27:
@garethadams One more question, which of the installs did you try?

#566 -> `npm install "git://github.com/ghostoy/node-serialport.git#nan1to2"`
or 
#571 -> `npm install "git://github.com/julianduque/node-serialport.git#master"`

garethadams, 2015-09-27:
Yes, it was a virtual serial port hooked up to a terminal and all it did was echo the input to the console but it worked.

I tried: #566 ->  `npm install "git://github.com/ghostoy/node-serialport.git#nan1to2"`

fivdi, 2015-09-27:
@garethadams ok, thanks :)

cpustar, 2015-09-27:
@fivdi 

Solved it. 

I was having difficulty installing it, node-gyp wasn't working properly because I had more than one version of VS installed.

fivdi, 2015-09-27:
@cpustar that's good news :), thanks for the feedback.

Quick question, which of the installs did you try out and were you able to talk to a device?

gnail, 2015-09-27:
I've been trying to get node-serialport to run under 4.1.1 and I'm getting access violations with both ghostoy's and julianduque's code:

ghostoy's branch:
```
npm ERR! Windows_NT 6.1.7601
npm ERR! argv "C:\\Program Files (x86)\\nodejs\\node.exe" "C:\\Users\\---\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "start"
npm ERR! node v4.1.1
npm ERR! npm  v3.3.3
npm ERR! code ELIFECYCLE
npm ERR! ---@--- start: `node --harmony ./bin/www`
npm ERR! Exit status 3221225477
```
Attaching node.exe to VS2012 debugger reports
```
First-chance exception at 0x778C7090 (ntdll.dll) in node.exe: 0xC0000005: Access violation reading location 0xFFAB0018.
Unhandled exception at 0x778C7090 (ntdll.dll) in node.exe: 0xC0000005: Access violation reading location 0xFFAB0018.
The program '[2356] node.exe' has exited with code 0 (0x0).
```

julianduque's branch:
```
npm ERR! Windows_NT 6.1.7601
npm ERR! argv "C:\\Program Files (x86)\\nodejs\\node.exe" "C:\\Users\\---\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "start"
npm ERR! node v4.1.1
npm ERR! npm  v3.3.3
npm ERR! code ELIFECYCLE
npm ERR! ---@--- start: `node --harmony ./bin/www`
npm ERR! Exit status 3221225477
```
VS2012 debugger:
```
First-chance exception at 0x778C43E0 (ntdll.dll) in node.exe: 0xC0000005: Access violation reading location 0x00000004.
Unhandled exception at 0x778C43E0 (ntdll.dll) in node.exe: 0xC0000005: Access violation reading location 0x00000004.
```
I don't have debugging symbols setup for node/serialport so that's probably not very useful.

This does not happen on node 0.12.7 or io.js 2.5.0 with node-serialport 1.7.4.

I'm working on a minimal reproducible case at the moment but it seems like it's something to do with 'use strict', binary files, and long-running parser functions.

fivdi, 2015-09-27:
@gnail Were there lots compile errors? AFAIK VS2013 is needed to compile native addons for Node.js v4 on Windows.

gnail, 2015-09-27:
Ok, the 'use strict'/--harmony and the binary file thing were red herrings and are probably irrelevant.

I've created a [minimal test case](https://gist.github.com/gnail/a1a2c0d11205c00286ab) and I can recreate this ~70% of the time using either PRs.

My test setup uses FTDI [USB null-modem](http://www.ftdichip.com/Products/Cables/USBtoUSB.htm) cables with both ends plugged into my computer so it presents as two serial ports and the data loops back to the other port. 

I use [RealTerm](http://realterm.sourceforge.net/) to send binary data from one of the ports and open the other port with the gist linked above. For data files I was using a 7z archive as a source of (mostly) random binary file and a GitHub client log as a text file.

Usually node will exit on the third parser call but sometimes it can be the fourth as well. I have never seen it exit on any other invocations - if it goes past 4th invocation it seems to happily run indefinitely from then on. Running the gist with or without --harmony doesn't seem to make a statistically significant difference.

gnail, 2015-09-27:
@fivdi Not really, one deprecation warning and that's it. Here are the outputs for both PRs using x86 node.js 4.1.1 and MSVS2013 :

```
C:\Users\-------\-------\------------->npm install "https://github.com/ghostoy/n
ode-serialport.git#nan1to2"
npm WARN install Couldn't install optional dependency: Unsupported

> serialport@1.7.4 install C:\Users\-------\-------\-------------\node_modules\s
erialport
> node-pre-gyp install --fallback-to-build

Building the projects in this solution one at a time. To enable parallel build,
please add the "/m" switch.
  serialport.cpp
  serialport_win.cpp
  enumser.cpp
..\src\win\enumser.cpp(443): warning C4996: 'GetVersionExA': was declared depre
cated [C:\Users\-------\-------\-------------\node_modules\serialport\build\ser
ialport.vcxproj]
          C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433)
  : see declaration of 'GetVersionExA'
  disphelper.c
  win_delay_load_hook.c
     Creating library C:\Users\-------\-------\-------------\node_modules\seria
  lport\build\Release\serialport.lib and object C:\Users\-------\-------\------
  -------\node_modules\serialport\build\Release\serialport.exp
  Generating code
  Finished generating code
  serialport.vcxproj -> C:\Users\-------\-------\-------------\node_modules\ser
  ialport\build\Release\\serialport.node
  Copying C:\Users\-------\-------\-------------\node_modules\serialport\build\
  Release\/serialport.node to C:/Users/-------/-------/-------------/node_modul
  es/serialport/build/serialport/v1.7.4/Release/node-v46-win32-ia32\serialport.
  node
          1 file(s) copied.
------------@------ C:\Users\-------\-------\-------------
└── serialport@1.7.4  (git+https://github.com/ghostoy/node-serialport.git#09658d
ff63d12f7932ef4b06fb7aaa791d48b1de)
```

```
C:\Users\-------\-------\------------->npm install "https://github.com/julianduq
ue/node-serialport.git#master"
npm WARN install Couldn't install optional dependency: Unsupported

> serialport@1.7.4 install C:\Users\-------\-------\-------------\node_modules\s
erialport
> node-pre-gyp install --fallback-to-build

Building the projects in this solution one at a time. To enable parallel build,
please add the "/m" switch.
  serialport.cpp
  serialport_win.cpp
  enumser.cpp
..\src\win\enumser.cpp(443): warning C4996: 'GetVersionExA': was declared depre
cated [C:\Users\-------\-------\-------------\node_modules\serialport\build\ser
ialport.vcxproj]
          C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433)
  : see declaration of 'GetVersionExA'
  disphelper.c
  win_delay_load_hook.c
     Creating library C:\Users\-------\-------\-------------\node_modules\seria
  lport\build\Release\serialport.lib and object C:\Users\-------\-------\------
  -------\node_modules\serialport\build\Release\serialport.exp
  Generating code
  Finished generating code
  serialport.vcxproj -> C:\Users\-------\-------\-------------\node_modules\ser
  ialport\build\Release\\serialport.node
  Copying C:\Users\-------\-------\-------------\node_modules\serialport\build\
  Release\/serialport.node to C:/Users/-------/-------/-------------/node_modul
  es/serialport/build/serialport/v1.7.4/Release/node-v46-win32-ia32\serialport.
  node
          1 file(s) copied.
------------@------ C:\Users\-------\-------\-------------
└── serialport@1.7.4  (git+https://github.com/julianduque/node-serialport.git#84
7fe0cb0eafc163bd2e74b47f17e8e1e5dce39d)
```

cpustar, 2015-09-28:
@fivdi 

I used #566 -> npm install "git://github.com/ghostoy/node-serialport.git#nan1to2"

and it works fine.

fivdi, 2015-09-28:
@gnail it looks like you have uncovered an issue as you're seeing different behavior with Node.js 0.12.7/io.js 2.5.0 and Node.js 4.1.1.

fivdi, 2015-09-28:
@cpustar ok, thanks for the feedback.

tigoe, 2015-09-28:
I ran both branches on OSX, both worked flawlessly. The test case referenced above has some unnneeded dependencies, but like @gnail I wrote my own, and it worked fine. Haven't been able to set up a win box yet, though, but working on it. Useful to see the errors in others, though.

fivdi, 2015-09-28:
@tigoe Nice, nice. This makes me happy.

shaunEll, 2015-09-29:
Finally got it to install (Windows 10x64, VS2015, Node 4.1.1) after installing c++ for VS2015

Used this #556 npm install "git://github.com/ghostoy/node-serialport.git#nan1to2"

Will test tomorrow

Thanks all

fivdi, 2015-09-29:
@shaunEll that's good news. Please let us know how the tests go.

shaunEll, 2015-09-29:
Hi, tested this morning and it seems to be working fine (i'm a node noob). I listed my available ports and received data with no issues. Haven't tried writing yet, i need to update my Arduino sketch 1st.

Is there anything else that i should try out?

fivdi, 2015-09-29:
@shaunEll you could try writing with serialport with your own Arduino sketch.
Or you could try this Arduino test: https://github.com/voodootikigod/node-serialport/tree/master/arduinoTest
Or you could try either with Node.js 0.10.40 or 0.12.7, ...

fivdi, 2015-09-29:
@shaunEll BTW, thanks for the feedback :)

fivdi, 2015-09-29:
Everything looks fine with [Raspbian Jessie](https://www.raspberrypi.org/downloads/raspbian/) which was released recently and ships with gcc/g++ 4.9.2. #566 installs successfully with Node.js 4.1.1 and a few simple read/write tests ran immaculately.

MAOShea, 2015-09-29:
@shaunEll , you're a star!

shaunEll, 2015-09-29:
@MAOShea , Glad to help :)

@fivdi  I just checked and serialport is writing fine to the arduino (with node 4.1.1)

fivdi, 2015-09-29:
@shaunEll Excellent, thanks!

jacobrosenthal, 2015-09-30:
Thanks and keep feedback coming everyone. Looks like #566 builds fro everyone, but there are some [possible runtime issues on windows](https://github.com/voodootikigod/node-serialport/issues/578#issuecomment-143591608) uncovered by @gnail

tigoe, 2015-09-30:
I set up a virgin Windows 10 install and installed node 4.11, gitHub for Windows, and Python 2.7. After much mucking about with paths, I tried to install #566 as above. Current error looks like this:

```
MSBUILD : error MSB4132: The tools version "2.0" is unrecognized. Available tools versions are "4.0".
gyp ERR! build error
gyp ERR! stack Error: `msbuild` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Windows_NT 10.0.10240
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=C:\\Users\\tigoe\\Documents\\nodeser\\node_modules\\serialport\\build\\serialport\\v1.7.4\\Release\\node-v46-win32-x64\\serialport.node" "--module_name=serialport" "--module_path=C:\\Users\\tigoe\\Documents\\nodeser\\node_modules\\serialport\\build\\serialport\\v1.7.4\\Release\\node-v46-win32-x64"
gyp ERR! cwd C:\Users\tigoe\Documents\nodeser\node_modules\serialport
gyp ERR! node -v v4.1.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd build --fallback-to-build --module=C:\Users\tigoe\Documents\nodeser\node_modules\serialport\build\serialport\v1.7.4\Release\node-v46-win32-x64\serialport.node --module_name=serialport --module_path=C:\Users\tigoe\Documents\nodeser\node_modules\serialport\build\serialport\v1.7.4\Release\node-v46-win32-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\tigoe\Documents\nodeser\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:817:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Windows_NT 10.0.10240
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\tigoe\\Documents\\nodeser\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\tigoe\Documents\nodeser\node_modules\serialport
node-pre-gyp ERR! node -v v4.1.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.11
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd build --fallback-to-build --module=C:\Users\tigoe\Documents\nodeser\node_modules\serialport\build\serialport\v1.7.4\Release\node-v46-win32-x64\serialport.node --module_name=serialport --module_path=C:\Users\tigoe\Documents\nodeser\node_modules\serialport\build\serialport\v1.7.4\Release\node-v46-win32-x64' (1)
npm ERR! Windows_NT 10.0.10240
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "git://github.com/ghostoy/node-serialport.git#nan1to2"
npm ERR! node v4.1.1
npm ERR! npm  v2.14.4
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\tigoe\Documents\nodeser\npm-debug.log
```

Not sure if it's the particular pull request or if my node toolchain's still not complete. will do a bit more digging, and try the other pull request as well.

tigoe, 2015-09-30:
Same failure as above with #571

shaunEll, 2015-09-30:
@tigoe What version of Visual Studio are you running?

tigoe, 2015-09-30:
V2015, Freshly installed. Being a windows newbie for programming, I'm having a hell of a time setting the build path. I keep getting:

```
C:\Users\tigoe\Documents\nodeser\node_modules\serialport\build\serialport.vcxproj(20,3): error MSB4019: The imported project "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V110\Microsoft.Cpp.Default.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
```

I'm missing something obvious yet undocumented, right? Any clues?

shaunEll, 2015-09-30:
Have you checked to see if that msbuild\microsoft.cpp\v4.0\v110 path exists? Mine is v140 for the same setup as yours. 
Did you have to set the *VCTargetsPath* variable manually?

tigoe, 2015-09-30:
Tried setting a symlink from V110 to V140 in the v4.0 directory, and I set the VCTargetsPath variable, but still got a failure. But I didn't try them together, I'll give that a go, thanks.

MAOShea, 2015-09-30:
I think your command line is missing --msvs_version=xxxx where xxxx = the
year of your Visual Studio. In my case, with VS2015 Community Edition, it's
--msvs_version=2015

Hope this helps.

Michael

Jackbennett, 2015-09-30:
Windows 8.1 x64 with node v4.1.1 and VS Community 2015 installed. Did file -> new project -> C++ just to ensure the files are included. I'm sure there's some way to just get the C++ and MS build tools without the whole VS suite. Been a long time since I needed to compile this project.

`npm install "git://github.com/ghostoy/node-serialport.git#nan1to2" --msvs_version=2015` was all I needed to get it to build.

fivdi, 2015-09-30:
@Jackbennett were you able to talk to some type of device with Node v4.1.1 and serialport after the successful install?

tigoe, 2015-10-01:
@Jackbennett's post gave me hope, so I uninstalled node.js and Visual Studio, then reinstalled them, and I was finally able to get it to compile, #566 Then I ran a few serial tests with an Arduino. Working fine so far, I've gotten serial data going both directions at 9600bps.

Testing conditions:
* Windows 10, 64 bit system, using node.js 4.1.1 and Visual Studio Community Edition 2015. 
* Installed serialport using the same line as Jack, above.
* Tested using these examples: https://github.com/tigoe/NodeExamples/tree/master/SerialIntro
* On the Arduino side, used Arduino 1.6.5, using an Uno and the AnalogReadSerial and PhysicalPixel examples from the examples included with the IDE.

I'll do a run at 115200, to make sure it works at  top speed.

tigoe, 2015-10-01:
Running the ArduinoEcho sketch with a simplified version of stress.js (without the dependencies) at 115200, I find it always stops running at 852 bytes on Windows. Not sure why, but it's consistent. I'll try to submit a pull request with my version of the stress test.

mgcrea, 2015-10-01:
@tigoe That's good news. I think we should not wait too much feedback before bumping a new major, as old setup will still work on the current major (thanks semver), and a lot of new feedback will arrive once there is an official tag out there, after what, releasing a couple of patch versions to address theses new issues should not be problematic.

Thanks! Serialport is one of the last pre-4.0 deps that are holding our team back, we're pretty excited to finally move forward.

fivdi, 2015-10-01:
>> as old setup will still work on the current major (thanks semver)

@mgcrea could you elaborate a little on this statement please? Are you suggesting that serialport v1.7.4 will work on Node.js v4.1.1? If so, this is not the case.

fivdi, 2015-10-01:
@gnail I don't have a Windows machine or a USB null modem cable but attempted to simulate the issue you're having with a Raspberry Pi 2 and an ATmega328P. Although the parser in the [minimal test case](https://gist.github.com/gnail/a1a2c0d11205c00286ab) is quite synthetic in that it simply outputs 10000 lines to the console every time it's passed a buffer of data, it still looks like it should work. The test ran successfully many times without error. This is the [code that was used on the Raspberry Pi 2](https://gist.github.com/fivdi/36632ab930a71ba62c16) and this is the [code that was used on the ATmega328P](https://gist.github.com/fivdi/adaa00fa3e7eccfaa4cd).

mgcrea, 2015-10-01:
@fivdi I was just saying that any existing working (thus pre node-v4) setup requiring `serialport@^1.7` would not be broken by a new major release (as long as they don't link against `latest` or `*`).

Major releases are meant to break things so while I think it's good to check that things are not completly broken on any platforms, delaying a working release by weeks to iron some platform-specific bugs would not be the best thing to do.

[57 more replies]

closed 2015-10-21 by jacobrosenthal