serialport/node-serialport #579: Raspberry pi install issues

jacobrosenthal, 2015-09-12:
Hoping to confirm or deny #535 #489 #469 #409 #368
any others?

In the readme we say
```
   wget http://nodejs.org/dist/v0.10.16/node-v0.10.16-linux-arm-pi.tar.gz
   tar xvfz node-v0.10.16-linux-arm-pi.tar.gz
   sudo mv node-v0.10.16-linux-arm-pi /opt/node/
```

Ive just now successfully tested a rapsberry pi 2 which has node v0.10.40 (via nodesources link)
```
curl -sL https://deb.nodesource.com/setup | sudo bash
```

Is there a reason our docs link to a specific node version. Should we just change that line and close these?

nebrius, 2015-09-12:
Reposting from gitter, for posterity:

In my experience, the official node-arm builds were really spotty back when joyent still held the reigns, we would sometimes go long periods w/o arm builds, so we (raspi-io/johnny-five) switched to this other build.

There was always the `latest` file, so we wouldn't have to update docs either, which is nice.

It was also nice having .deb packages, since we could then install with apt-get and have it handle PATHing

I would guess that’s why serialport linked to a specific version, but I couldn’t say for certain

jacobrosenthal, 2015-09-14:
closed (for now) by #580

jacobrosenthal, 2015-09-14:
OOps. Well that was hasty apparently. latest points at 4.0

https://node-arm.herokuapp.com/
NOTE: You will have to upgrade to jessie version of debian in order to run latest node!
Replace all instances of "wheezy" in /etc/apt/sources.list

Not sure thatll work @nebrius

The argument against node source seemed to be it wasnt a deb. Looks like it is these days... and we can lock to node 10
curl -sL https://deb.nodesource.com/setup_0.10 | sudo -E bash -

https://github.com/nodesource/distributions

Thoughts?

nebrius, 2015-09-14:
And now I has a sad :( Looks like we have to lock to a specific version after all.

Should we lock to node 0.10 or node 0.12? I would say 0.12, but that's simply my own personal bias since I run 0.12 everywhere.

jacobrosenthal, 2015-09-14:
I personally dont have an opinion, weve been hard coding 10 releases. Smallest change now is to lock to 10 latest.

fivdi, 2015-09-14:
If it's possible to lock to 0.10 or 0.12 this implies that both 0.10 and 0.12 function correctly. If both versions function correctly it isn't actually necessary to lock.

Why is the locking necessary?

jacobrosenthal, 2015-09-14:
@fivdi This is around the install instructions which are hardcoding a node 10 version, but then never get updated leading to people who follow them to a t opening saying they can't install. We're just picking a working latest branch we're comfortable linking to.

jacobrosenthal, 2015-09-14:
Also of note, the windows instructions still point a 10 installer.

nebrius, 2015-09-14:
@fivdi, we know we can't point to node 4 (yet), so we have to pick a "not latest" version of node, in effect locking it to a prior version. I was hoping the node-arm site wasn't going to update so fast, but oh well.

fivdi, 2015-09-14:
The same binaries are available at [https://github.com/nathanjohnson320/node_arm/tree/master/routes/files](https://github.com/nathanjohnson320/node_arm/tree/master/routes/files), so using
```
wget https://github.com/nathanjohnson320/node_arm/blob/master/routes/files/node_0.12.6-1_armhf.deb
```
instead of
```
wget http://node-arm.herokuapp.com/node_latest_armhf.deb
```
should work.

A similar issue was opened at nathanjohnson320/node_arm today: https://github.com/nathanjohnson320/node_arm/issues/12

jacobrosenthal, 2015-09-14:
Thats what we were doing (hard coding a specific point release), were
trying to not do that.

fivdi, 2015-09-14:
That may have been the intention initially, but the following comments led me to believe that the plan had changed

>Should we lock to node 0.10 or node 0.12? I would say 0.12, but that's simply my own personal bias since I run 0.12 everywhere.

>I personally dont have an opinion, weve been hard coding 10 releases. Smallest change now is to lock to 10 latest.

I don't think it's going to be possible to find a "working latest branch" that will work with serialport as latest implies v4, so the following won't work:

>We're just picking a working latest branch we're comfortable linking to.

jacobrosenthal, 2015-09-14:
Were all set, node source hosts a 10, 12 and 4 latest

On Mon, Sep 14, 2015 at 11:36 PM, Brian Cooke <notifications@github.com>
wrote:

> go

fivdi, 2015-09-14:
Yes, but for the Pi 2 which is and ARMv7 machine and not for the Pi 1 which is an ARMv6 machine.

fivdi, 2015-09-14:
Linking to NodeSource for Pi 2 users is definitely a good idea though.

jacobrosenthal, 2015-09-14:
Sigh, yes that appears true.

fivdi, 2015-09-14:
How about linking to NodeSource for Pi 2 users and linking to https://github.com/nathanjohnson320/node_arm/blob/master/routes/files/node_0.12.6-1_armhf.deb for Pi 1 users?

jacobrosenthal, 2015-09-15:
@nebrius  do you have any relationship with whomever does node-arm, can you ask for a 10, 12, 4 latest?

nebrius, 2015-09-15:
@jacobrosenthal I don't have a relationship with them, but they keep all the old archives online, e.g. http://node-arm.herokuapp.com/node_0.10.36_armhf.deb.

N4TY, 2015-09-15:
Hope I'm not intruding here (or piling on), but I'm trying to install Node-Red along with serialport on RPI-B model 1.  I'm using a Jessie release found here:  http://gnutoolchains.com/raspberry/jessie/

This works:
 wget http://node-arm.herokuapp.com/node_latest_armhf.deb
 sudo dpkg -i node_latest_armhf.deb
 sudo apt-get install build-essential python-dev python-rpi.gpio
 sudo npm install -g node-red

AND:
root@pi:~# node --version
v4.0.0
root@pi:~# npm --version
2.14.2
root@pi:~# python --version
Python 2.7.9

And THEN get to serialport:
root@pi:~# npm install serialport

> serialport@1.7.4 install /root/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/root/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.h:5:0,
                 from ../src/serialport.cpp:3:
../node_modules/nan/nan.h:261:25: error: redefinition of ‘template<class T> v8::Local<T> _NanEnsureLocal(v8::Local<T>)’
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                         ^
../node_modules/nan/nan.h:256:25: note: ‘template<class T> v8::Local<T> _NanEnsureLocal(v8::Handle<T>)’ previously declared here
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                         ^
../node_modules/nan/nan.h:661:13: error: ‘node::smalloc’ has not been declared
     , node::smalloc::FreeCallback callback
             ^
../node_modules/nan/nan.h:661:35: error: expected ‘,’ or ‘...’ before ‘callback’
     , node::smalloc::FreeCallback callback
                                   ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(char*, size_t, int)’:
../node_modules/nan/nan.h:665:50: error: ‘callback’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                  ^
../node_modules/nan/nan.h:665:60: error: ‘hint’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                            ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(const char*, uint32_t)’:
../node_modules/nan/nan.h:672:67: error: call of overloaded ‘New(v8::Isolate*, const char*&, uint32_t&)’ is ambiguous
     return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
                                                                   ^
../node_modules/nan/nan.h:672:67: note: candidates are:
In file included from ../node_modules/nan/nan.h:25:0,
                 from ../src/serialport.h:5,
                 from ../src/serialport.cpp:3:
/root/.node-gyp/4.0.0/include/node/node_buffer.h:31:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, v8::Local<v8::String>, node::encoding) <near match>
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/root/.node-gyp/4.0.0/include/node/node_buffer.h:31:40: note:   no known conversion for argument 3 from ‘uint32_t {aka unsigned int}’ to ‘node::encoding’
/root/.node-gyp/4.0.0/include/node/node_buffer.h:43:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t) <near match>
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/root/.node-gyp/4.0.0/include/node/node_buffer.h:43:40: note:   no known conversion for argument 2 from ‘const char*’ to ‘char*’
In file included from ../src/serialport.h:5:0,
                 from ../src/serialport.cpp:3:
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(uint32_t)’:
../node_modules/nan/nan.h:676:61: error: could not convert ‘node::Buffer::New(v8::Isolate::GetCurrent(), size)’ from ‘v8::MaybeLocal<v8::Object>’ to ‘v8::Local<v8::Object>’
     return node::Buffer::New(v8::Isolate::GetCurrent(), size);
                                                             ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanBufferUse(char*, uint32_t)’:
../node_modules/nan/nan.h:683:12: error: ‘Use’ is not a member of ‘node::Buffer’
     return node::Buffer::Use(v8::Isolate::GetCurrent(), data, size);
[... 25 lines cut ...]
node-pre-gyp ERR! cwd /root/node_modules/serialport
node-pre-gyp ERR! node -v v4.0.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/root/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-linux-arm/serialport.node --module_name=serialport --module_path=/root/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-linux-arm' (1)
npm ERR! Linux 4.1.6+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport" "--save"
npm ERR! node v4.0.0
npm ERR! npm  v2.14.2
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/npm-debug.log

keyvanfatehi, 2015-09-15:
Here has been my experience for the record

I use node-serialport in on the BeagleBone Black which is armv7 hardfloat.

I was originally using node 0.12.6 which compiles node-serialport fine. Don't use it though.

There is a memory leak that will show up more you use node-serialport under that node version. I was able to track this down by using pm2 & keymetrics.io and watching memory graph over a 24 hour period while sending data over rs-232. This leak does NOT happen in node of the same version on x86, so it was hard to believe until I saw it.

Ever since switching to iojs 2.5.0 that leak is gone. The process uses way less memory in a default state (it actually uses 25-50% less memory just as a fresh process), and is way more stable overall.

I couldn't get node-serialport to compile on any version beyond that, so I am using iojs 2.5.0 now everywhere I use node-serialport.

**tl;dr; use iojs 2.5.0**

```
wget https://iojs.org/dist/v2.5.0/iojs-v2.5.0-linux-armv7l.tar.gz
tar -C /usr/local --strip-components 1 -xzf iojs-v2.5.0-linux-armv7l.tar.gz
```

torgeir, 2015-09-20:
Stumbled on this myself, trying to build serialport w/node v4.1 on the pi 2

jacobrosenthal, 2015-09-30:
@nebrius when I looked node-arm (now?) has a node 12 branch called archive so I just pulled in c05cdfef3274036633348a2cf9ac5006bf8438c8 . This doesnt solve @keyvanfatehi's memory leak report, well have to figure that out next

closed 2015-09-30 by jacobrosenthal

torgeir, 2015-11-07:
For the record I got this working in the end by doing: https://gist.github.com/torgeir/89ec54bbc6cf80d8bff4