serialport/node-serialport #587: Disconnecting serial cable crashes node

lizhouatforte, 2015-09-16:
on node 0.10.36 with serial port 1.7.4 on windows 7 64bit

If I run node with serial cable plugged, then unplug the cable, the node is crashed.

Here is my code:

var serialport = require("serialport");
var sp = null;
serialport.list(function (err, ports) {
            ports.some(function (port) {
                sp = new serialport.SerialPort(port.comName, { baudrate: 19200 });
                sp.on('open', function () {
                    console.log('Opened ' + port.comName);
                });
                sp.on('error', function () {
                    console.log('Errored!!!');
                });
                sp.on('close', function () {
                    console.log('Closed!!!');
                });
                return true;
            });
        });

The console output is:

Opened COM3

undefined:0
TypeError: undefined is not a function

I appreciate any input why error/close events are not triggered.  Thanks!

reconbot, 2016-03-28:
I figured out why, this is bananas. Adding this to the backlog as `disconnect` is not documented or working properly.

reconbot, 2016-03-28:
I figured out why, this is bananas. Adding this to the backlog as `disconnect` is not documented or working properly. #638

reconbot, 2016-04-06:
Disconnect still removes all events but it does it after emitting `disconnect` and `error` events. We're going to stop doing this behavior in the next major release. 

We fixed a ton of bugs in our latest beta specifically ones addressing this issue. https://github.com/voodootikigod/node-serialport/issues/733 Please try it out `serialport@2.0.7-beta3` and report back!

reconbot, 2016-04-10:
The CPU issue seems fixed but the port can't be reopened. Still an issue.

reconbot, 2016-09-02:
Per #919 window's serialport poller will emit errors on disconnect that can't be caught.

reconbot, 2017-07-08:
This is fixed with 5.0.0-beta6

closed 2017-07-08 by reconbot