serialport/node-serialport #590: Confusion about supported versions of io.js and Node.js

fivdi, 2015-09-20:
Over the last week or two I've seen some confusion about what versions of io.js and Node.js are supported by node-serialport. Perhaps some of the confusion is related to the [How To Use](https://github.com/voodootikigod/node-serialport/tree/v1.7.4#how-to-use) and [To Install](https://github.com/voodootikigod/node-serialport/tree/v1.7.4#to-install) sections of the readme. Both these sections of the readme are probably a little dated and updating them is likely to help prevent confusion. For example, would it not be better to remove the information about v0.8, v0.10, and v0.11 and say something like "node-serialport supports Node.js v0.10 through v0.12 and io.js v1.0 through v2.5." (or whatever the correct version numbers are)?

jacobrosenthal, 2015-09-30:
Agreed, but sadly I dont know what [this](https://github.com/voodootikigod/node-serialport/blame/master/README.md#L60-L67) refers to either. Perhaps if @voodootikigod remembers he can let us know what ticket or platform that might have referred to and someone can do some tests.

Otherwise, Im ruthless about support so I'm just as happy saying 11.x isnt supported.

Likewise, Im not too sure about which if any io releases should be supported.

fivdi, 2015-09-30:
One portion of the text linked to is:

>Currently support for Node.js version 0.11.x is dealing with an issue in the latest version of v. 0.11.13. We have confirmed things are fine with 0.11.10 and earlier, but not 0.11.11+.

Although this was true when written in August 2014, it's incorrect today. It's not possible for serialport to work fine with 0.11.10 and earlier as [nan removed support for Node 0.11 versions prior to 0.11.13](https://github.com/nodejs/nan/tree/v1.5.3#may-2014-major-changes-for-v8-325--node-01113)

I can't think of a sane reason for supporting 11.x today, so dropping support for it sounds reasonable to me.

voodootikigod, 2015-10-11:
We should not support 0.11.x period.

jacobrosenthal, 2015-10-11:
Clarified in https://github.com/voodootikigod/node-serialport/commit/51fac6ce630552e7c87631bcd21efac5e6f7df02

closed 2015-10-11 by jacobrosenthal