serialport/node-serialport #591: Node 4.0

desvdp, 2015-09-21:
I've seen quite some posts on supporting or not supporting Node 4.0. 
Would it be possible to clarify the "official" state ... I tend to believe there's a lot of confusion right now, maybe that's just me ...

jacobrosenthal, 2015-09-21:
See https://github.com/voodootikigod/node-serialport/issues/578

closed 2015-09-21 by jacobrosenthal

desvdp, 2015-09-22:
Hi Jacob, an honest question (but potentially stupid question, if so please excuse me): reading your posts on this site and gitter it seems many of the issues boil down to the need for gcc 4.8. What I can't find is what drives this need, i.e. I'm running node 4.1.0 without issues on a mint RPi2 which, at least as far I can tell, only has gcc-4.6.3 (default wheezy image, did do a dist-update but without modifying the source file to jessie).
What am I missing? Is this a requirement for the upgrade to nan2 (I can't find it) ... 
Thanks for clarifying...

FYI: I installed node using the download image found on the official node.js site.