serialport/node-serialport #592: Building fails with node v0.11.10

vicatcu, 2015-09-21:
I'm trying to find a stable combination of node and node-serialport and I'm struggling. Based on the README.md statement "We have confirmed things are fine with 0.11.10" that's not consistent with my experience. I'm on Ubuntu 14.04 64-bit, and I'm using nvm to manage node versions. With that configuration, the first few errors I hit are:

```
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from /home/vic/.node-gyp/0.11.10/src/node.h:61:0,
                 from ../node_modules/nan/nan.h:24,
                 from ../src/serialport.h:5,
                 from ../src/serialport.cpp:3:
/home/vic/.node-gyp/0.11.10/deps/v8/include/v8.h: In member function ‘void v8::ReturnValue<T>::Set(uint32_t)’:
/home/vic/.node-gyp/0.11.10/deps/v8/include/v8.h:5816:31: warning: typedef ‘I’ locally defined but not used [-Wunused-local-typedefs]
   typedef internal::Internals I;
                               ^
In file included from ../src/serialport.h:5:0,
                 from ../src/serialport.cpp:3:
../node_modules/nan/nan.h: At global scope:
../node_modules/nan/nan.h:62:9: error: ‘UnboundScript’ in namespace ‘v8’ does not name a type
 typedef v8::UnboundScript      NanUnboundScript;
         ^
```

fivdi, 2015-09-22:
>"We have confirmed things are fine with 0.11.10"

At the time of writing this statement was correct, however, it's no longer true. node-serialport depends on nan, but [nan no longer supports Node 0.11 versions prior to 0.11.13](https://github.com/nodejs/nan/tree/v1.5.3#may-2014-major-changes-for-v8-325--node-01113).

Try with Node 0.12.7.

vicatcu, 2015-09-22:
Yes works with 0.12.7

fivdi, 2015-09-22:
That's good news :)

fivdi, 2015-09-24:
@vicatcu can this issue be closed?

vicatcu, 2015-09-24:
I think it's a documentation only problem. README.md is not accurate in
what the latest stable release is.

Victor Aprea // Wicked Device
victor.aprea@wickeddevice.com
T: @vicatcu <http://twitter.com/vicatcu/>

fivdi, 2015-09-24:
There's another issue for fixing the documentation https://github.com/voodootikigod/node-serialport/issues/590

jacobrosenthal, 2015-10-11:
Frankly have no idea whats up with 0.11.x, but it gets cloudier as we move to nan2. Recommend 10, 12, or node4 as our support of it evolves.

closed 2015-10-11 by jacobrosenthal