serialport/node-serialport #594: I have the same issue

ericaslakson, 2015-09-24:
node-pre-gyp ERR! cwd C:\Users\btl0\Downloads\nwjs-sdk-v0.13.0-alpha3-win-x64\nwjs-sdk-v0.13.0-alpha3-win-x64\node_modules\serialport
node-pre-gyp ERR! node -v v4.1.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd build --fallback-to-build --module=C:\Users\btl0\Downloads\nwjs-sdk-v0.13.0-alpha3-win-x64\nwjs-sdk-v0.13.0-alpha3-win-x64\node_modules\serialport\build\serialport\v1.7.4\Release\node-v46-win32-x64\serialport.node --module_name=serialport --module_path=C:\Users\btl0\Downloads\nwjs-sdk-v0.13.0-alpha3-win-x64\nwjs-sdk-v0.13.0-alpha3-win-x64\node_modules\serialport\build\serialport\v1.7.4\Release\node-v46-win32-x64' (1)
npm ERR! Windows_NT 6.1.7601
npm ERR! argv "C:\\Program Files (x86)\\nodejs\\node.exe" "C:\\Program Files (x86)\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "i" "serialport"
npm ERR! node v4.1.1
npm ERR! npm  v2.14.4
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\btl0\Downloads\nwjs-sdk-v0.13.0-alpha3-win-x64\nwjs-sdk-v0.13.0-alpha3-win-x64\npm-debug.log

fivdi, 2015-09-24:
@ericaslakson I see that you're on a windows machine :+1: 

As can be seen [here](https://github.com/voodootikigod/node-serialport/issues/578), node-serialport is looking for Windows testers for the pull requests that add support for Node.js 4.

There are currently two pull requests for Node.js 4 support, https://github.com/voodootikigod/node-serialport/pull/566 and https://github.com/voodootikigod/node-serialport/pull/571.

https://github.com/voodootikigod/node-serialport/pull/566 can be installed with npm `install "git://github.com/ghostoy/node-serialport.git#nan1to2"`

https://github.com/voodootikigod/node-serialport/pull/571 can be installed with `npm install "git://github.com/julianduque/node-serialport.git#master"`

You could try either or both of them on Windows and report your findings :+1:

jacobrosenthal, 2015-09-30:
@ericaslakson as fvdi said, Please try #566 and let us know how it goes in #578

closed 2015-09-30 by jacobrosenthal