serialport/node-serialport #595: Build Failed with OSX10.9.5 and Ubuntu14.04?

weiway, 2015-09-24:
Here's the log:
```
Weis-MacBook-Pro:AppChromerino Wei$ npm install serialport
npm WARN package.json app-chromerino@0.1.0 No repository field.
npm WARN package.json app-chromerino@0.1.0 No README data
-
> serialport@1.7.4 install /Users/Wei/Xproject/AppChromerino/node_modules/serialport
> node-pre-gyp install --fallback-to-build

  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../node_modules/nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../node_modules/nan/nan.h:661:13: error: no member named 'smalloc' in namespace
      'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/Wei/.node-gyp/4.1.1/include/node/node_buffer.h:31:40: note: candidate
      function not viable: no known conversion from 'uint32_t'
      (aka 'unsigned int') to 'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/Wei/.node-gyp/4.1.1/include/node/node_buffer.h:43:40: note: candidate
      function not viable: 2nd argument ('const char *') would lose const
      qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/Wei/.node-gyp/4.1.1/include/node/node_buffer.h:28:40: note: candidate
      function not viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/Wei/.node-gyp/4.1.1/include/node/node_buffer.h:36:40: note: candidate
      function not viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:676:12: error: no viable conversion from
      'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/Wei/.node-gyp/4.1.1/include/node/v8.h:210:7: note: candidate constructor
      (the implicit copy constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'const v8::Local<v8::Object> &' for 1st
      argument
class Local {
      ^
/Users/Wei/.node-gyp/4.1.1/include/node/v8.h:210:7: note: candidate constructor
      (the implicit move constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&' for 1st
      argument
class Local {
      ^
/Users/Wei/.node-gyp/4.1.1/include/node/v8.h:214:13: note: candidate template
      ignored: could not match 'Local' against 'MaybeLocal'
  V8_INLINE Local(Local<S> that)
            ^
/Users/Wei/.node-gyp/4.1.1/include/node/v8.h:326:13: note: candidate template
      ignored: could not match 'S *' against 'v8::MaybeLocal<v8::Object>'
  V8_INLINE Local(S* that)
            ^
In file included from ../src/serialport.cpp:3:
[... 106 lines cut ...]
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/Wei/Xproject/AppChromerino/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64/serialport.node --module_name=serialport --module_path=/Users/Wei/Xproject/AppChromerino/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64' (1)
npm ERR! Darwin 13.4.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport"
npm ERR! node v4.1.1
npm ERR! npm  v2.14.4
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/Wei/Xproject/AppChromerino/npm-debug.log
Weis-MacBook-Pro:AppChromerino Wei$ 
```

fivdi, 2015-09-24:
node-serialport doesn't support Node.js v4.1.1 yet. See https://github.com/voodootikigod/node-serialport/issues/578 for further information.

There are currently two pull requests for Node.js 4 support, https://github.com/voodootikigod/node-serialport/pull/566 and https://github.com/voodootikigod/node-serialport/pull/571.

https://github.com/voodootikigod/node-serialport/pull/566 can be installed with `npm install "git://github.com/ghostoy/node-serialport.git#nan1to2"`

https://github.com/voodootikigod/node-serialport/pull/571 can be installed with `npm install "git://github.com/julianduque/node-serialport.git#master"`

You could try either or both of them and report your findings :+1:

weiway, 2015-09-25:
Both #566 and #571 build succeed on my OSX10.9.5 with node v4.1.1, did not test on ubuntu. Thank you!

jacobrosenthal, 2015-09-26:
Thank you for the confirmations

On Sat, Sep 26, 2015 at 12:34 AM, Wei Wang <notifications@github.com> wrote:

> 5 with node v4

LeandroFavero, 2015-09-27:
Not work in Raspberry Pi2 :-(

fivdi, 2015-09-27:
@LeandroFavero what issues have you seen on the Raspberry Pi 2?

tmladek, 2015-09-29:
Neither works on me on Arch Linux :(

Logs: https://gist.github.com/tmladek/eba28cb6f7ff3033120f

kammce, 2015-09-29:
Both #566 and #571 builds succeed on machine. Using Ubuntu 14.04 with IO.js 3.2.0. Will be upgrading to nodejs 4.x.x to to test this further. Will be testing the port on real hardware soon.

fivdi, 2015-09-29:
@kammce thanks for the feedback :)
Please let us know how the tests go.

kammce, 2015-09-29:
Continuing to use io.js 3.2.0 on ubuntu 14.04LTS, both #566 and #571 work to read and write information to an arduino. 

Nodejs Program:

var SerialPort = require("serialport").SerialPort
var serialPort = new SerialPort("/dev/ttyACM0", {
  baudrate: 9600
});

serialPort.on("open", function () {
        console.log('open');
        serialPort.on('data', function(data) {
                console.log('data received: ' + data);
        });
        setInterval(function() {
                serialPort.write("hello serial port\n");
        }, 1000);
});

used example Arduino SerialEvent.

fivdi, 2015-09-29:
@kammce Excellent :+1:

kammce, 2015-09-29:
And finally, I can confirm that with nodejs v4.1.1 with the same setup both #566 and #571 it works.

fivdi, 2015-09-29:
@kammce Even better :+1:

closed 2015-09-30 by jacobrosenthal

jacobrosenthal, 2015-09-30:
Thanks for all the reports, closing this for the main node 4 ticket.