serialport/node-serialport #597: How can I use node-serialport in Windows XP?

Lanfei, 2015-09-28:
I ran `npm install serialport` in Windows XP, but failed. Here is the log:

```
D:\serialPort\node_modules\serialport>if not defined npm_config_node_gyp (node "
C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\no
de-gyp\bin\node-gyp.js" clean )  else (node  clean )

D:\serialPort\node_modules\serialport>if not defined npm_config_node_gyp (node "
C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\no
de-gyp\bin\node-gyp.js" configure --fallback-to-build --module=D:\serialPort\nod
e_modules\serialport\build\serialport\v1.7.4\Release\node-v46-win32-ia32\serialp
ort.node --module_name=serialport --module_path=D:\serialPort\node_modules\seria
lport\build\serialport\v1.7.4\Release\node-v46-win32-ia32 )  else (node  configu
re --fallback-to-build --module=D:\serialPort\node_modules\serialport\build\seri
alport\v1.7.4\Release\node-v46-win32-ia32\serialport.node --module_name=serialpo
rt --module_path=D:\serialPort\node_modules\serialport\build\serialport\v1.7.4\R
elease\node-v46-win32-ia32 )

D:\serialPort\node_modules\serialport>if not defined npm_config_node_gyp (node "
C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\no
de-gyp\bin\node-gyp.js" build --fallback-to-build --module=D:\serialPort\node_mo
dules\serialport\build\serialport\v1.7.4\Release\node-v46-win32-ia32\serialport.
node --module_name=serialport --module_path=D:\serialPort\node_modules\serialpor
t\build\serialport\v1.7.4\Release\node-v46-win32-ia32 )  else (node  build --fal
lback-to-build --module=D:\serialPort\node_modules\serialport\build\serialport\v
1.7.4\Release\node-v46-win32-ia32\serialport.node --module_name=serialport --mod
ule_path=D:\serialPort\node_modules\serialport\build\serialport\v1.7.4\Release\n
ode-v46-win32-ia32 )
MSBUILD : error MSB4132: The tools version "2.0" is unrecognized.
gyp ERR! build error
gyp ERR! stack Error: `C:\WINDOWS\Microsoft.NET\Framework\v4.0.30319\msbuild.exe
` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\
npm\node_modules\node-gyp\lib\build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_proces
s.js:200:12)
gyp ERR! System Windows_NT 5.1.2600
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodej
s\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallb
ack-to-build" "--module=D:\\serialPort\\node_modules\\serialport\\build\\serialp
ort\\v1.7.4\\Release\\node-v46-win32-ia32\\serialport.node" "--module_name=seria
lport" "--module_path=D:\\serialPort\\node_modules\\serialport\\build\\serialpor
t\\v1.7.4\\Release\\node-v46-win32-ia32"
gyp ERR! cwd D:\serialPort\node_modules\serialport
gyp ERR! node -v v4.1.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd build --fallback-
to-build --module=D:\serialPort\node_modules\serialport\build\serialport\v1.7.4\
Release\node-v46-win32-ia32\serialport.node --module_name=serialport --module_pa
th=D:\serialPort\node_modules\serialport\build\serialport\v1.7.4\Release\node-v4
6-win32-ia32' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (D:\serialPort\node_modu
les\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:817:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/chi
ld_process.js:211:5)
node-pre-gyp ERR! System Windows_NT 5.1.2600
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\serialPort\
\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "insta
ll" "--fallback-to-build"
node-pre-gyp ERR! cwd D:\serialPort\node_modules\serialport
node-pre-gyp ERR! node -v v4.1.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd build --fallback-to-build --module=D:\serialPort
\node_modules\serialport\build\serialport\v1.7.4\Release\node-v46-win32-ia32\ser
ialport.node --module_name=serialport --module_path=D:\serialPort\node_modules\s
erialport\build\serialport\v1.7.4\Release\node-v46-win32-ia32' (1)
npm ERR! Windows_NT 5.1.2600
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\
node_modules\\npm\\bin\\npm-cli.js" "install" "serialport" "--registry=https://r
egistry.npm.taobao.org"
npm ERR! node v4.1.1
npm ERR! npm  v2.14.4
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --f
allback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     D:\serialPort\npm-debug.log
```

jedthehumanoid, 2015-10-16:
What version of Visual Studio do you have installed?

bhaveshgohel, 2015-10-16:
@Lanfei  : Can you try using `` npm install "git://github.com/voodootikigod/node-serialport.git#master" `` ?

closed 2015-10-19 by jacobrosenthal

Lanfei, 2015-11-08:
Thanks you all, I have installed `node-serialport` successfully by using:
```
npm install serialport --msvs_version=2012
```