serialport/node-serialport #601: Problem with raspberry Pi

bloudman, 2015-10-01:
Hi, i've got a problem with my raspberry pi (all my script works perfectly on windows)

On my rpy I've got this error:

gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: read ECONNRESET
gyp ERR! stack     at exports._errnoException (util.js:746:11)
gyp ERR! stack     at TCP.onread (net.js:559:26)
gyp ERR! System Linux 4.1.7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/home/pi/alarme/node_modules/serialport/build/serialport/v1.7.4/Release/node-v14-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/pi/alarme/node_modules/serialport/build/serialport/v1.7.4/Release/node-v14-linux-arm"
gyp ERR! cwd /home/pi/alarme/node_modules/serialport
gyp ERR! node -v v0.12.6
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/pi/alarme/node_modules/serialport/build/serialport/v1.7.4/Release/node-v14-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/alarme/node_modules/serialport/build/serialport/v1.7.4/Release/node-v14-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/alarme/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:1015:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1087:5)
node-pre-gyp ERR! System Linux 4.1.7+
node-pre-gyp ERR! command "node" "/home/pi/alarme/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/alarme/node_modules/serialport
node-pre-gyp ERR! node -v v0.12.6
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/pi/alarme/node_modules/serialport/build/serialport/v1.7.4/Release/node-v14-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/alarme/node_modules/serialport/build/serialport/v1.7.4/Release/node-v14-linux-arm' (1)
npm ERR! Linux 4.1.7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "update"
npm ERR! node v0.12.6
npm ERR! npm  v2.11.2
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

fivdi, 2015-10-01:
> gyp ERR! stack Error: read ECONNRESET

The Raspberry Pi is having network problems.

bloudman, 2015-10-02:
okay, thanks for your help!
I made some changement on my rpi  and now i got this:
pi@alarme ~/alarme $ sudo npm install serialport --unsafe-perm                                                                
npm WARN package.json NodejsConsoleApp2@0.0.0 No repository field.                                                            
npm WARN package.json NodejsConsoleApp2@0.0.0 No README data                                                                  
npm WARN package.json http@0.0.0 http is also the name of a node core module.                                                 
npm WARN package.json http@0.0.0 No description                                                                               
npm WARN package.json http@0.0.0 No repository field.                                                                         
npm WARN package.json http@0.0.0 No README data                                                                               
|                                                                                                                             
> serialport@1.7.4 install /home/pi/alarme/node_modules/serialport                                                            
> node-pre-gyp install --fallback-to-build                                                                                    
                                                                                                                              
gyp WARN install got an error, rolling back install                                                                           
gyp ERR! configure error                                                                                                      
gyp ERR! stack Error: node-v0.12.0.tar.gz local checksum 2f739097e31c404aa9ca79e831c12b3f32bfd974b3dac09fc4fc5ec6faa14caf not 
match remote 9700e23af4e9b3643af48cef5f2ad20a1331ff531a12154eef2bfb0bb1682e32                                                 
gyp ERR! stack     at deref (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:299:20)                     
gyp ERR! stack     at IncomingMessage.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:340:13
)                                                                                                                             
gyp ERR! stack     at IncomingMessage.emit (events.js:129:20)                                                                 
gyp ERR! stack     at _stream_readable.js:908:16                                                                              
gyp ERR! stack     at process._tickCallback (node.js:355:11)                                                                  
gyp ERR! System Linux 4.1.7+                                                                                                  
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--
fallback-to-build" "--module=/home/pi/alarme/node_modules/serialport/build/serialport/v1.7.4/Release/node-v14-linux-arm/serial
port.node" "--module_name=serialport" "--module_path=/home/pi/alarme/node_modules/serialport/build/serialport/v1.7.4/Release/n
ode-v14-linux-arm"                                                                                                            
gyp ERR! cwd /home/pi/alarme/node_modules/serialport                                                                          
gyp ERR! node -v v0.12.0                                                                                                      
gyp ERR! node-gyp -v v1.0.2                                                                                                   
gyp ERR! not ok                                                                                                               
node-pre-gyp ERR! build error                                                                                                 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bi
n/node-gyp.js configure --fallback-to-build --module=/home/pi/alarme/node_modules/serialport/build/serialport/v1.7.4/Release/n
ode-v14-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/alarme/node_modules/serialport/build/serialp
ort/v1.7.4/Release/node-v14-linux-arm' (1)                                                                                    
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/alarme/node_modules/serialport/node_modules/node-pre-gyp/lib
/util/compile.js:83:29)                                                                                                       
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)                                                           
node-pre-gyp ERR! stack     at maybeClose (child_process.js:1008:16)                                                          
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1080:5)                                  
node-pre-gyp ERR! System Linux 4.1.7+                                                                                         
node-pre-gyp ERR! command "node" "/home/pi/alarme/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallbac
[... cut ...]
npm ERR!     npm owner ls serialport                                                                                          
npm ERR! There is likely additional logging output above.                                                                     
                                                                                                                              
npm ERR! Please include the following file with any support request:                                                          
npm ERR!     /home/pi/alarme/npm-debug.log

fivdi, 2015-10-02:
Things are going terribly wrong but I don't know why.

The following error indicates that there may be connectivity issues:

>gyp ERR! stack Error: node-v0.12.0.tar.gz local checksum 2f739097e31c404aa9ca79e831c12b3f32bfd974b3dac09fc4fc5ec6faa14caf not
match remote 9700e23af4e9b3643af48cef5f2ad20a1331ff531a12154eef2bfb0bb1682e32

A github search for `NodejsConsoleApp2` from the following warning only returns hits related to Windows:

>npm WARN package.json NodejsConsoleApp2@0.0.0 No repository field.

The issue isn't related to serialport. I'd imagine that similar issues will occur when other native modules are installed, for example, `npm install epoll`.

Are you behind a firewall?
Is the OS a standard Raspbian distribution or something else?

bloudman, 2015-10-02:
I'm not behind a fierewall and it's the standard os (and the node version for arm version 6..)

I dont' know what happened ^^ 

I think i will take an old real PC for my project! :)

thanks a lot for your help guy !

fivdi, 2015-10-03:
@bloudman Are you using Wi-Fi on the Raspberry Pi? If so, you could try with an Ethernet cable instead.

jacobrosenthal, 2015-10-11:
Feel free to continue conversion, but Im going to close as an active issue.

closed 2015-10-11 by jacobrosenthal