serialport/node-serialport #603: Not able to install on Mac OSX El Capitan

dmongeau, 2015-10-03:
Hi,
Since I updated to El Capitan, i'm not able to install `node-serialport`. I get the following errors:

```
> serialport@1.7.4 install /Users/dmp/Code/gifomathon/ecran/node_modules/serialport
> node-pre-gyp install --fallback-to-build

  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../../nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../../nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../../nan/nan.h:661:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../../nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/dmp/.node-gyp/4.1.1/src/node_buffer.h:31:40: note: candidate function not viable: no known
      conversion from 'uint32_t' (aka 'unsigned int') to 'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/dmp/.node-gyp/4.1.1/src/node_buffer.h:43:40: note: candidate function not viable: 2nd argument
      ('const char *') would lose const qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/dmp/.node-gyp/4.1.1/src/node_buffer.h:28:40: note: candidate function not viable: requires 2
      arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/dmp/.node-gyp/4.1.1/src/node_buffer.h:36:40: note: candidate function not viable: requires 5
      arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../../nan/nan.h:676:12: error: no viable conversion from 'v8::MaybeLocal<v8::Object>' to
      'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/dmp/.node-gyp/4.1.1/deps/v8/include/v8.h:210:7: note: candidate constructor
      (the implicit copy constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'const v8::Local<v8::Object> &' for 1st argument
class Local {
      ^
/Users/dmp/.node-gyp/4.1.1/deps/v8/include/v8.h:210:7: note: candidate constructor
      (the implicit move constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&' for 1st argument
class Local {
      ^
/Users/dmp/.node-gyp/4.1.1/deps/v8/include/v8.h:214:13: note: candidate template ignored: could not
      match 'Local' against 'MaybeLocal'
  V8_INLINE Local(Local<S> that)
            ^
/Users/dmp/.node-gyp/4.1.1/deps/v8/include/v8.h:326:13: note: candidate template ignored: could not
      match 'S *' against 'v8::MaybeLocal<v8::Object>'
  V8_INLINE Local(S* that)
            ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../../nan/nan.h:683:26: error: no member named 'Use' in namespace 'node::Buffer'
    return node::Buffer::Use(v8::Isolate::GetCurrent(), data, size);
           ~~~~~~~~~~~~~~^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
In file included from ../../nan/nan.h:24:
[... 93 lines cut ...]
node-pre-gyp ERR! node -v v4.1.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/Cellar/node/4.1.1/bin/node /Users/dmp/Code/gifomathon/ecran/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/dmp/Code/gifomathon/ecran/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64/serialport.node --module_name=serialport --module_path=/Users/dmp/Code/gifomathon/ecran/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64' (1)
npm ERR! Darwin 15.0.0
npm ERR! argv "/usr/local/Cellar/node/4.1.1/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v4.1.1
npm ERR! npm  v3.3.5
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/dmp/Code/gifomathon/ecran/npm-debug.log
```

dtex, 2015-10-03:
Hi @dmongeau, I don't think this is an El Capitan thing. Node v4.x support is not ready yet. Follow #578 to see what's going on with that. In there you will find two forks that add support for node 4.x. You can install one of those and use node 4.x or you could roll back to node 0.12.7 and all should be good.

I'm not sure how much testing has been done against El Capitan so if you could report back on how the above solutions work out, that would certainly be helpful.

dmongeau, 2015-10-03:
Yes, sorry about that, just switched to node v0.12.7 and it's working

closed 2015-10-03 by dmongeau

jacobrosenthal, 2015-10-03:
@dmongeau <https://github.com/dmongeau> can you confirm if this is still
true in the official release?
https://github.com/voodootikigod/node-serialport/issues/552