serialport/node-serialport #605: Does node-serialport not fully "claim" the seriaport?

tigoe, 2015-10-07:
We've been using node-serialport extensively with P5.js on OSX lately (Yosemite, node v.0.12),and I noticed something really strange. node-serialport doesn't stop other programs from accessing the serial port when it's got the port open. But it does check to see if another program has the port open before it tries to open the port. So if you open the port through node-serialport and then open it with another program (in my case, the Arduino serial monitor), you can have the port open in two applications, but neither one will be able to read all the bytes.

It causes some inconsistent behavior. For example, sometimes you can upload a new Arduino sketch while node has the port open, and sometimes you can't. Unplugging the serial device is the only surefire way of clearing all claims on the serial port.

Used to be ports had to write to the lockfile to claim the port. Is that no longer the case?  How is it that node-serialport can open a port and still allow other programs to access the port?

durkie, 2015-11-02:
Yep, I'm also seeing this -- connected in my node app, and able to open up a separate serial connection to the port and talk to my arduino. 

I think this could be solved with the flock(2) call, or applying the O_EXLOCK flag in the open(2) call that opens the serial port, but I haven't yet been able to build a version that tests this.

reconbot, 2016-04-04:
We lock on windows (by chance or design) but not on osx or linux.

reconbot, 2016-05-16:
closing for #810

closed 2016-05-16 by reconbot