serialport/node-serialport #606: Can't Install Serialport on El Capitan

CraigMcCahill, 2015-10-09:
Hi I'm tying to work with Johnny Five and Arduino on a Mac running El Capitan (10.11) - but can't install Serial Port. Node is installed via Homebrew. 

Npm version is:

    npm -version
    2.14.6

Node version is:

    node -v
    v4.1.2

xCode Command Line Tools are installed. Apologies if I've done something obviously wrong :)

Full error text is below:- 
 
```
Craigs-MacBook-Pro:~ craigmccahill$ npm install serialport
> serialport@1.7.4 install /Users/craigmccahill/node_modules/serialport
> node-pre-gyp install --fallback-to-build

 CXX(target) Release/obj.target/serialport/src/serialport.o
 In file included from ../src/serialport.cpp:3:
 In file included from ../src/serialport.h:5:
 ../node_modules/nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../node_modules/nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../node_modules/nan/nan.h:661:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/craigmccahill/.node-gyp/4.1.2/include/node/node_buffer.h:31:40: note: candidate function not viable: no known conversion from 'uint32_t' (aka 'unsigned int') to
      'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/craigmccahill/.node-gyp/4.1.2/include/node/node_buffer.h:43:40: note: candidate function not viable: 2nd argument ('const char *') would lose const qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/craigmccahill/.node-gyp/4.1.2/include/node/node_buffer.h:28:40: note: candidate function not viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/craigmccahill/.node-gyp/4.1.2/include/node/node_buffer.h:36:40: note: candidate function not viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:676:12: error: no viable conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/craigmccahill/.node-gyp/4.1.2/include/node/v8.h:210:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'const v8::Local<v8::Object> &' for 1st argument
class Local {
      ^
/Users/craigmccahill/.node-gyp/4.1.2/include/node/v8.h:210:7: note: candidate constructor (the implicit move constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&' for 1st argument
class Local {
      ^
/Users/craigmccahill/.node-gyp/4.1.2/include/node/v8.h:214:13: note: candidate template ignored: could not match 'Local' against 'MaybeLocal'
  V8_INLINE Local(Local<S> that)
            ^
/Users/craigmccahill/.node-gyp/4.1.2/include/node/v8.h:326:13: note: candidate template ignored: could not match 'S *' against 'v8::MaybeLocal<v8::Object>'
  V8_INLINE Local(S* that)
            ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
[... 98 lines cut ...]
npm ERR! npm  v2.14.6
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/craigmccahill/npm-debug.log`

```

Great library if I can't getting working on the Mac I'll give it a go on a PC tomorrow.

Thanks 

Craig

dtex, 2015-10-09:
Hi @CraigMcCahill , El Capitan is not the culprit. You'll want to track this issue: #578 

In that issure there are two forks of Node Serialport you can try out with node.js 4.x or you can roll back to node.js 0.12.x and it should work fine.

rwaldron, 2015-10-09:
@jacobrosenthal this can be closed as a duplicate

closed 2015-10-13 by jacobrosenthal