serialport/node-serialport #608: Error: No compatible version found - Raspberry Pi

alexcroox, 2015-10-12:
Trying to install on a Raspberry Pi 2 (Raspbian OS updated, but rolled back to node v0.10.2 until 4.x is supported).
```
node -v
v0.10.2
```

```
npm -v
1.2.15
```

```
npm install --save serialport
npm http GET https://registry.npmjs.org/serialport
npm http 304 https://registry.npmjs.org/serialport
npm ERR! Error: No compatible version found: serialport@'^1.5.0'
npm ERR! Valid install targets:
npm ERR! ["0.1.0","0.1.1","0.1.3","0.2.0","0.2.2","0.2.3","0.2.4","0.2.5","0.2.6","0.2.7","0.2.8","0.2.9","0.3.0","0.6.0","0.6.1","0.6.2","0.6.3","0.6.5","0.6.6","0.6.7","0.7.0","0.7.1","0.7.2","0.7.3","0.7.4","0.7.5","1.0.0","1.0.1","1.0.2","1.0.3","1.0.4","1.0.5","1.0.6","1.0.7","1.0.8","1.1.0","1.1.1","1.1.3","1.2.0","1.2.1","1.2.2","1.2.3","1.2.4","1.2.5","1.3.0","1.3.1","1.4.0","1.4.1","1.4.2","1.4.5","1.4.6","1.4.8","1.4.9","1.4.10","1.5.0","1.6.1","1.6.2","1.6.3","1.7.0","1.7.1","1.7.3","1.7.4"]
npm ERR!     at installTargetsError (/usr/local/lib/node_modules/npm/lib/cache.js:685:10)
npm ERR!     at /usr/local/lib/node_modules/npm/lib/cache.js:607:10
npm ERR!     at saved (/usr/local/lib/node_modules/npm/node_modules/npm-registry-client/lib/get.js:138:7)
npm ERR!     at Object.oncomplete (fs.js:107:15)
npm ERR! If you need help, you may report this log at:
npm ERR!     <http://github.com/isaacs/npm/issues>
npm ERR! or email it to:
npm ERR!     <npm-@googlegroups.com>

npm ERR! System Linux 4.1.7-v7+
npm ERR! command "/usr/local/bin/node" "/usr/local/bin/npm" "install" "--save" "serialport"
npm ERR! cwd /home/pi
npm ERR! node -v v0.10.2
npm ERR! npm -v 1.2.15
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/npm-debug.log
npm ERR! not ok code 0
```

fivdi, 2015-10-12:
Not sure if anyone is going to be able to help with Node.js 0.10.2 as it's so old. There have been **38** v0.10.x releases since v0.10.2. I don't know if NAN (one of serialports dependencies) actually supports v0.10.2 either.

alexcroox, 2015-10-12:
Yer I was on 4.2 but this library doesn't support it so I rolled back :)

fivdi, 2015-10-12:
Ok, go back to 4.2 and try this: `npm install "git://github.com/voodootikigod/node-serialport.git#master"`

fivdi, 2015-10-12:
Master should work, it just hasn't been published yet.

fivdi, 2015-10-12:
It was just published, so `npm install serialport` should work with Node.js 4.2 now.

jacobrosenthal, 2015-10-13:
Please reopen if you're having trouble still.

closed 2015-10-13 by jacobrosenthal