serialport/node-serialport #609: Application aborted without errors.

luizlago, 2015-10-13:
I use serialport for gcode injection on my 3dprinter. In nodejs 0.12.7 everything works well, but node 4.x in random moment my app abort without any error. I disabled the serialport and nothing aborted. When I enable the serialport the problem appears.

In the screenshot ... the printer is warming up to begin printing, the app was aborted. 

[image]

Anybody know whats happening? How appear any error?

jacobrosenthal, 2015-10-13:
There were (at least) two issues found in the node 4 upgrade that have yet to be corroborated or chased down. Perhaps bit by one of these?
* https://github.com/voodootikigod/node-serialport/issues/578#issuecomment-143591608
with [this test](https://gist.github.com/gnail/a1a2c0d11205c00286ab)
* https://github.com/voodootikigod/node-serialport/issues/578#issuecomment-144756809
with [this test](https://github.com/voodootikigod/node-serialport/pull/600/files)

ValentinCupif, 2015-10-20:
Same issue on Linux. it comes when i closing COM.

nicozhang, 2015-10-28:
Same issue. Anyone can solve this problem?

luizlago, 2015-10-29:
I solved my problem between serialport and electron using this fork ... https://github.com/ddm/node-serialport.

jacobrosenthal, 2015-11-09:
Can you all try 2.0.4 which has a windows patch that might fix this?

closed 2015-11-12 by luizlago