serialport/node-serialport #610: Linux (especially embedded) serial port listing issues

jacobrosenthal, 2015-10-13:
We've got problems around listing ports, especially around using /dev/serial 

https://github.com/voodootikigod/node-serialport/pull/576/
https://github.com/voodootikigod/node-serialport/pull/561
https://github.com/voodootikigod/node-serialport/pull/565

jacobrosenthal, 2015-10-21:
@xuqianli @lfk @abrinsmead Can you all `npm install https://github.com/jacobrosenthal/node-serialport/tree/mitsos1os-master` to see if it solves your port listing issues or creates any new issues

lfk, 2015-10-21:
@jacobrosenthal Thank you for summarizing these issues. I should have access to some suitable hardware tonight or tomorrow -- I'll check the proposed solution and provide feedback ASAP. As for my PR, it was simply a quick fix that wouldn't have impacted backwards compatibility at all. Naturally I'm all for finding a "better" way of doing things, and as long as feature parity can be assured, I've got nothing against moving away from the udev-dependency. Cheers!

jacobrosenthal, 2015-10-22:
@digihaven Can you all npm install https://github.com/jacobrosenthal/node-serialport/tree/mitsos1os-master to see if it solves your port listing issues or creates any new issues

jacobrosenthal, 2015-10-24:
Added ttyS and ttyAMA, Can anyone provide any feedback on their linux boxes, especially around identical devices. Also Can anyone dig up any more /dev/ files we should be whitelisting?

bigmonkeyboy, 2015-10-24:
The popular Beaglebone Black board uses  ttyO
eg see - http://www.armhf.com/beaglebone-black-serial-uart-device-tree-overlays-for-ubuntu-and-debian-wheezy-tty01-tty02-tty04-tty05-dtbo-files/

jacobrosenthal, 2015-11-09:
@techninja apparently you've found problems with the new linux port listing that came as of 2.0.3?  Id love to hear about which platforms and what it looks like.

techninja, 2015-11-09:
@jacobrosenthal Yes yes yes, I have. I've been spending today looking into figuring it out and seeing if it's duplicated in another ticket. Might not be. Just tried your `mitsos1os-master` branch to no avail.

Ubuntu 14.04: Port listing dies with 64 "Uncaught TypeError: Cannot read property 'split' of undefined" from line 541 of serialport.js. It's looking for `DEVLINKS` in the object, but it's just not there. Here's the first one it gets: 
```
DEVNAME: "/dev/ttyS8"
DEVPATH: "/devices/platform/serial8250/tty/ttyS8"
ID_MM_CANDIDATE: "1"
MAJOR: "4"
MINOR: "72"
SUBSYSTEM: "tty"
USEC_INITIALIZED: "6557"
```
My guess is that these are just system TTYs not hardware serial ports like we expect to be listing. I [obviously blame](https://github.com/voodootikigod/node-serialport/commit/87d3fa08a73bd9a619dc85d16ea22f646a16831a) @reconbot ;)

jacobrosenthal, 2015-11-09:
@techninja  This is 2.0.4?

techninja, 2015-11-09:
Actually was using #623 directly (which you merged a bit ago). You do a `npm publish` for 2.0.4 since then?

jacobrosenthal, 2015-11-09:
Yes

techninja, 2015-11-09:
Don't answer that, I did. Just installed latest 2.0.4 and getting the same issue. So, the answer is yes definitely. There a command I can run to list the same TTYs on the console? I'm generally suspecting filtering is the issue here.

fivdi, 2015-11-09:
I'm seeing the same error with serialport v2.0.4 on a Raspberry Pi 2
```
pi@raspberrypi ~/dev/sp $ node -v
v5.0.0

pi@raspberrypi ~/dev/sp $ cat list.js 
var serialPort = require("serialport");
serialPort.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(port);
    console.log(port.comName);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});

pi@raspberrypi ~/dev/sp $ node list.js 
/home/pi/dev/sp/node_modules/serialport/serialport.js:541
      var pnpId = as_json.DEVLINKS.split(' ')[0];
                                  ^

TypeError: Cannot read property 'split' of undefined
    at udev_parser (/home/pi/dev/sp/node_modules/serialport/serialport.js:541:35)
    at /home/pi/dev/sp/node_modules/serialport/serialport.js:592:11
    at ChildProcess.exithandler (child_process.js:194:7)
    at emitTwo (events.js:87:13)
    at ChildProcess.emit (events.js:172:7)
    at maybeClose (internal/child_process.js:818:16)
    at Socket.<anonymous> (internal/child_process.js:319:11)
    at emitOne (events.js:77:13)
    at Socket.emit (events.js:169:7)
    at Pipe._onclose (net.js:469:12)
pi@raspberrypi ~/dev/sp $ 
```

jacobrosenthal, 2015-11-09:
:( Sigh. The list code is
https://github.com/voodootikigod/node-serialport/blob/master/serialport.js#L575

Whats your distro, node, npm etc?

techninja, 2015-11-09:
Node 4.2.1 stable, npm 2.14.7, same error in Electron as well (which is as we know, a slightly different node env), shouldn't matter though.

Here's a full list of my system TTYs with one real one (the EBB controller shows up like the Arduino as `ttyACM0`):
```
tty
tty0
tty1
tty10
tty11
tty12
tty13
tty14
tty15
tty16
tty17
tty18
tty19
tty2
tty20
tty21
tty22
tty23
tty24
tty25
tty26
tty27
tty28
tty29
tty3
tty30
tty31
tty32
tty33
tty34
tty35
tty36
tty37
tty38
tty39
tty4
tty40
tty41
tty42
tty43
tty44
tty45
tty46
tty47
tty48
tty49
tty5
tty50
tty51
tty52
tty53
tty54
tty55
tty56
tty57
tty58
tty59
tty6
tty60
tty61
tty62
tty63
tty7
tty8
tty9
ttyACM0
ttyprintk
ttyS0
ttyS1
ttyS10
ttyS11
ttyS12
ttyS13
ttyS14
ttyS15
ttyS16
ttyS17
ttyS18
ttyS19
ttyS2
ttyS20
ttyS21
ttyS22
ttyS23
ttyS24
ttyS25
ttyS26
ttyS27
ttyS28
ttyS29
ttyS3
ttyS30
ttyS31
ttyS4
ttyS5
ttyS6
ttyS7
ttyS8
ttyS9
 ```

fivdi, 2015-11-09:
On the Raspberry Pi `/dev/ttyAMA0` is actually detected. `as_json` [here](https://github.com/voodootikigod/node-serialport/blob/master/serialport.js#L540) has the following value when the error is thrown:
```
{ DEVNAME: '/dev/ttyAMA0',
  DEVPATH: '/devices/platform/soc/3f201000.uart/tty/ttyAMA0',
  MAJOR: '204',
  MINOR: '64',
  SUBSYSTEM: 'tty',
  TAGS: ':systemd:',
  USEC_INITIALIZED: '99425' }
```

node v5.0.0, npm v3.3.6, Raspbian Jessie 2015-09-24

```
pi@raspberrypi ~/dev/sp $ uname -a
Linux raspberrypi 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux
```

kmos, 2015-11-10:
I have a similar error on a Gentoo with the device on ```/dev/ACM0 ```:

```shell  
mos@snake ~/Projects/sport $ cat list.js
var serialPort = require("serialport");

serialPort.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(port.comName);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});
mos@snake ~/Projects/sport $ sudo node list.js
/home/mos/Projects/sport/list.js:5
    console.log(port.comName);
                    ^

TypeError: Cannot read property 'comName' of undefined
    at /home/mos/Projects/sport/list.js:5:21
    at Array.forEach (native)
    at /home/mos/Projects/sport/list.js:4:9
    at /home/mos/Projects/sport/node_modules/async/lib/async.js:254:17
    at done (/home/mos/Projects/sport/node_modules/async/lib/async.js:129:15)
    at /home/mos/Projects/sport/node_modules/async/lib/async.js:32:16
    at /home/mos/Projects/sport/node_modules/async/lib/async.js:251:21
    at /home/mos/Projects/sport/node_modules/serialport/serialport.js:585:15
    at ChildProcess.exithandler (child_process.js:210:5)
    at emitTwo (events.js:87:13)
mos@snake ~/Projects/sport $ node -v
v5.0.0
mos@snake ~/Projects/sport $ npm -v
3.3.6
mos@snake ~/Projects/sport $ uname -a
Linux snake 4.0.5-gentoo #1 SMP Sat Jul 18 13:27:50 CEST 2015 x86_64 Intel(R) Core(TM) i5-2520M CPU @ 2.50GHz GenuineIntel GNU/Linux ```

techninja, 2015-11-10:
Wow, lots of people running v5, and I thought it was crazy to jump on such a fast moving train. Hopefully Jacob has enough info to get moving on this. I moved to 2.0.1 for Mavericks support which works on Linux, but have to move to 2.04 for the Windows crash fix, which now doesn't work on Linux because we're finding too many of the wrong kind. Bleh!

garlinto, 2015-11-11:
I was having this install issue as well. I recently updated my rpi to Linux kernel version 4.1.7+ and was noticing numerous make (gcc) errors. After numerous attempts to install serialport, including previous versions, I began hunting around for a communication alternative such as I2C. In the process, I learned that the installed versions of gcc and g++ may be the culprit. After following the upgrade instructions [here](https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons) I was finally able to compile serialport@2.0.4 on my raspberry pi. Not sure if this fixes all of the compile errors noted here but it fixed mine.

fivdi, 2015-11-12:
Things have improved with serialport v2.0.5. Here's what can be seen on a Raspbery Pi 2 running Raspbian Jessie:

```
pi@raspberrypi ~/dev/serialport $ node -v
v5.0.0

pi@raspberrypi ~/dev/serialport $ cat list.js 
var serialPort = require("serialport");
serialPort.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(port);
    console.log(port.comName);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});

pi@raspberrypi ~/dev/serialport $ node list.js 
{ comName: '/dev/ttyAMA0',
  manufacturer: undefined,
  serialNumber: undefined,
  pnpId: undefined,
  vendorId: '0xundefined',
  productId: '0xundefined' }
/dev/ttyAMA0
undefined
undefined
pi@raspberrypi ~/dev/serialport $ 
```
The values of `vendorId` and `productId` don't look the best. `/dev/ttyAMA0` is a "real" serial port broken out to pins 8 and 10 on the GPIO header.

jacobrosenthal, 2015-11-22:
Vendor and Product id tracked separately here https://github.com/voodootikigod/node-serialport/issues/621
No noise since merging. Closing.

closed 2015-11-22 by jacobrosenthal