serialport/node-serialport #611: Windows

kant2002, 2015-10-14:
Installing `node-serialport` version 2.0 gives error on Windows

```
##### [master +0 ~1 -0]> npm install  serialport
npm WARN EPACKAGEJSON #####@1.0.0 No description
npm ERR! Windows_NT 6.1.7601
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\
node_modules\\npm\\bin\\npm-cli.js" "install" "serialport"
npm ERR! node v4.2.0
npm ERR! npm  v3.3.8
npm ERR! path #####\serialport\node_modules\node-p
re-gyp\node_modules\request\node_modules\har-validator\bin\har-validator
npm ERR! code ENOENT
npm ERR! errno -4058
npm ERR! syscall chmod

npm ERR! enoent ENOENT: no such file or directory, chmod '#####
\node_modules\serialport\node_modules\node-pre-gyp\node_modules\request\node_mod
ules\har-validator\bin\har-validator'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! Please include the following file with any support request:
npm ERR!     #####\npm-debug.log
```

domingosl, 2015-10-19:
I have the exact same problem on Windows 10 64bit.

C:\Users\Domingo\dev\hookninja>npm i serialport
npm WARN EPACKAGEJSON hookninja-config@0.1.0 No description
npm WARN EPACKAGEJSON hookninja-config@0.1.0 No repository field.
npm WARN EPACKAGEJSON hookninja-config@0.1.0 No license field.
npm ERR! Windows_NT 10.0.10240
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Domingo\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "i" "serialport"
npm ERR! node v4.2.1
npm ERR! npm  v3.3.8
npm ERR! path C:\Users\Domingo\dev\hookninja\node_modules\serialport\node_modules\node-pre-gyp\node_modules\request\node_modules\har-validator\bin\har-validator
npm ERR! code ENOENT
npm ERR! errno -4058
npm ERR! syscall chmod

npm ERR! enoent ENOENT: no such file or directory, chmod 'C:\Users\Domingo\dev\hookninja\node_modules\serialport\node_modules\node-pre-gyp\node_modules\request\node_modules\har-validator\bin\har-validator'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\Domingo\dev\hookninja\npm-debug.log

kant2002, 2015-10-20:
No, I resort to installing Node 0.12. My coworker mention that he has issues on x64, but not on x86 Node, so you could try that too.

kant2002, 2015-10-20:
That's likely related to `har-validator` or `request` module

jacobrosenthal, 2015-10-22:
I can't confirm this on serialport 2.0.1, windows 7 sp1, node 4.2.1, which is what I believe the original poster would be running.

kant2002, 2015-10-22:
what version of NPM do you have?

jacobrosenthal, 2015-10-22:
whatever comes with node 4.2.1, which appears to be 2.14.7

jacobrosenthal, 2015-10-22:
What I think I can confirm, is that I dont think binaries are working, and its going to want to do a local compile which will fail if you don't have all the pre-reqs (python 2.x, VS 2013) But that doesnt appear to be whats in the logs above

kant2002, 2015-10-22:
Before that compilation of branch which was intended to be port to NodeJS 4 was working fine. and I have all prerequisites on that PC to compile project. that's something different. It was a while so I don't exactly remember what I was trying to solve this issue

jacobrosenthal, 2015-10-22:
Node is incredibly unstable right now with the entire ecosystem in flux because of node4. Its possible har-validator or other packages were broken for a day in there and might be better now..

KonstantinBerezenko, 2015-10-22:
Same problem.  Windows 10, node v4.2.1, npm 3.3.8.

jacobrosenthal, 2015-11-09:
I think windows binaries are finally working. Please try the 2.0.4 release

closed 2015-11-12 by jacobrosenthal