serialport/node-serialport #612: ElCaptain 10.11 Running Node 4.2.1

djlaidler, 2015-10-18:
Hey Gents,

First, big thx to Jacob for replying to me in the wee hours of the morning... Legend!

Basically, I originally ran an install on Johnny Five on Node 4.0.0.
Then when attempting to run my first J5 script, I got;
Error: Cannot find module 'serialport'

Attempted to install serialport from;
npm install serialport

But it fails repeatedly... Even after Jacob's suggestion of upgrading Node to 4.2.1

Thoughts?

jacobrosenthal, 2015-10-18:
You emailed again and said you saw it complaining about not having xcode tools installed?

As we talked about you can run `xcode-select --install` to install them.  Did that work?

The question is why is the prebuilt failing to download?

djlaidler, 2015-10-18:
It seems I also had to upgrade npm, and for some strange reason, install Xcode.

Now I am getting a series of errors. Here is the first few lines;

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../../nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../../nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../../nan/nan.h:661:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../../nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/dan/.node-gyp/4.2.1/include/node/node_buffer.h:31:40: note: candidate function not viable: no known conversion from 'uint32_t' (aka 'unsigned int') to 'enum encoding' for 3rd
      argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/dan/.node-gyp/4.2.1/include/node/node_buffer.h:43:40: note: candidate function not viable: 2nd argument ('const char *') would lose const qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,

rwaldron, 2015-10-18:
Johnny-Five is not yet updated and it's not as simple as changing the dependency version.

jacobrosenthal, 2015-10-18:
Good to know, but It sounds like hes just attempting to install serialport
at this point?

djlaidler, 2015-10-18:
Johnny Five looks to install serialport 1.7.4 I believe, but this fails
also...

Should I raise this with rwaldron? Possibly he has experienced this issue.

_________________________
Daniel Laidler
djlaidler@gmail.com

jacobrosenthal, 2015-10-18:
sure

djlaidler, 2015-10-18:
Hey Guys,

Thx again for your feedback... I will cc you on the comms I have with
rwaldron..

Dan

_________________________
Daniel Laidler
djlaidler@gmail.com

rwaldron, 2015-10-19:
Why do we need another thread? sp 1.7.4 doesn't install on node 4. J5 doesn't yet know how to install the correct sp for the version of node it's running in. Expect release by tomorrow.

djlaidler, 2015-10-29:
Hey rwaldron, how did you get on? What is the latest stable versions of the johnny-five, serial port, node stack that we can work with. I got a new puter, and want to get something that works out of the box.
Thx in advance. Awesome work everyone.

rwaldron, 2015-10-29:
@djlaidler Everything is up to date now.

djlaidler, 2015-10-29:
Thx Rick, and SP guys. Keep on being awesome!

rwaldron, 2015-10-29:
@djlaidler ❤️

jacobrosenthal, 2015-11-09:
Presuming this is solved. Reopen if not

closed 2015-11-09 by jacobrosenthal