serialport/node-serialport #613: Initialization routine from dynamic link library (DLL) not completed

domingosl, 2015-10-18:
System:
Windows 10 64bit
SerialPort module compiled using VS2013 (no errors)

Upon var serialPort = require("serialport"), the console outputs:
Uncaught Error: Routine di inizializzazione della libreria di collegamento dinamico (DLL) non riuscita.
C:\Users\Domingo\dev\hookninja\node_modules\serialport\build\serialport\v2.0.1\Release\node-v46-win32-x64\serialport.node

Translated into English: Initialization routine from dynamic link library (DLL) not completed

Seems a problem in 64bit systems, any ideas?

dannystaple, 2015-10-26:
I have found that I am able to make this work starting using node, but it will not work if I start this with electron. Are you trying to use electron or similar with this?
I am also on a 64 bit system, but with serialport v2.0.2.

dannystaple, 2015-10-26:
Trying this: http://electron.atom.io/docs/v0.34.0/tutorial/using-native-node-modules/#the-easy-way - to see if that makes a difference... - doesn't work on windows. The other build options - I can't make them work on windows yet.

limbera, 2016-02-25:
We're experiencing similar symptoms. require("serialport") working fine on Node but when we try to use Electron we get the same error. Has anybody worked this out? Going to continue diving into gyp and pre-gyp today.

reconbot, 2016-03-31:
You can track electron support #538 as for the primary issue, @domingosl let us know if you're still having issues with the latest version (or beta) and I'm happy to help debug.

closed 2016-03-31 by reconbot