serialport/node-serialport #617: Versions 2.x crash with Windows 7 x64

ancasicolica, 2015-10-26:
Running my tool with serialport 1.7.4 with node 0.12.7 worked well. After upgrading node-serialport to 2.x, the application became instable, crashing very often (terminates without any notice, very randomly). After reverting to 1.7.4 everything was fine again. Same effect when running serialport with node 4 (only serialport 2.x).

No issues on Mac OS, only Windows is affected. As 1.7.4 is not node 4 compliant, this is a big issue for me.

jacobrosenthal, 2015-10-26:
Probably one of these:
* https://github.com/voodootikigod/node-serialport/issues/578#issuecomment-143591608
with [this test](https://gist.github.com/gnail/a1a2c0d11205c00286ab)
* https://github.com/voodootikigod/node-serialport/issues/578#issuecomment-144756809
with [this test](https://github.com/voodootikigod/node-serialport/pull/600/files)

ancasicolica, 2015-10-26:
Here is what I see when attaching the VS2015 debugger to the process (even it is not a lot):

Debug console:
`Critical error detected c0000374`

Call stack:

      ntdll.dll!RtlReportCriticalFailure()	Unbekannt
      ntdll.dll!RtlpReportHeapFailure()	Unbekannt
 	ntdll.dll!RtlpHeapHandleError()	Unbekannt
 	ntdll.dll!RtlpLogHeapFailure()	Unbekannt
 	ntdll.dll!RtlFreeHeap()	Unbekannt
 	kernel32.dll!HeapFree()	Unbekannt
 	node.exe!000000013ffffaac()	Unbekannt
 	node.exe!000000013fbad495()	Unbekannt
 	node.exe!000000013fbad15c()	Unbekannt
 	node.exe!000000013fbac29a()	Unbekannt
 	node.exe!000000013fbab561()	Unbekannt
 	node.exe!000000013fc55804()	Unbekannt
 	node.exe!000000013fb13939()	Unbekannt
 	node.exe!000000013fae0ed9()	Unbekannt
 	node.exe!000000013fd30950()	Unbekannt
 	node.exe!000000013fd3d990()	Unbekannt
 	node.exe!000000013fd3d828()	Unbekannt
 	node.exe!000000013fd5b04f()	Unbekannt
 	node.exe!000000013fd5b6ba()	Unbekannt
 	node.exe!000000013fd08de7()	Unbekannt`

livesankp, 2015-10-26:
Hello,

I have similar kind of issue. Trying to figure out from past 2 days if this is serialport or signalr issue but looks like serialport as my node program stops without any error or warning message while reading or writing to serialport. I have node.js v4.1.2 upgraded from 0.12 and serial port i tried 2.x and also this one npm install "git://github.com/ghostoy/node-serialport.git#nan1to2"  Both have similar issues i am able to write and read to serialport but suddenly it stops program. My program was working from past 6 months but not anymore after this upgrades.

nicozhang, 2015-10-29:
i have the same problem while using the code in reme

closed 2015-11-09 by jacobrosenthal

livesankp, 2015-11-11:
Looks liks this issue is fixed with node 4.2.2 and serialport 2.0.4

PR #623 (2015-11-09): Fix heap corruption issue on windows
In ```EIO_AfterWatchPort```, serialport creates a `v8::ArrayBuffer` using `Nan::NewBuffer() ` API which creates `v8::ArrayBuffer` with ` ArrayBufferCreationMode::kInternalized ` mode. This means that v8 engine takes the responsibility of freeing the external buffer passed to the `v8::ArrayBuffer`.
However ```EIO_AfterWatchPort``` deletes and frees the buffer at the end of method.
```C++
cleanup:
    delete data;
    delete req;
```
When ArrayBuffer object is garbage collected, it tries to free the external buffer passed to it, but since it is already freed, [... cut]