serialport/node-serialport #618: serialport is crashing node program without errors or warning

livesankp, 2015-10-28:
Hello I am using latest version of serial port v2.0.2.
Running node.js program which receives events from server and writes/read to serialport.
It was working fine on node v 0.12 until i upgraded node to 4.1.2 and serialport too.
My node program is working but suddenly it stops without any error or warning.
Couple of times i have noticed CONNRESET but i have a feeling that serialport is crashing somehow during read/write. Couple of times i was able to run without crash too. But it is mostly crashing.
I have tried "git://github.com/ghostoy/node-serialport.git#nan1to2" this too but same issue.

Are you aware of any such issue ? I am using windows x86 for testing.

nicozhang, 2015-10-29:
i meet the same problem

koralm, 2015-10-29:
Same, serialport 2.0.2 with node 4.2.1 crashes. When i start server without physically connected serialport (no transmision) everything is ok. But when i start transmission after 1-3s server is suddenly stop.

livesankp, 2015-10-30:
Yes you are right. It happens for me after 2 mins.
Able to caught this error using uncaughtException
process.on('uncaughtException', function(error) {
  console.error((new Date).toUTCString() + ' uncaughtException:', error.message)
  console.error(error.stack);
  process.exit(1)
})

Error is 

Fri, 30 Oct 2015 19:25:07 GMT uncaughtException: read ECONNRESET
Error: read ECONNRESET
    at exports._errnoException (util.js:849:11)
    at TCP.onread (net.js:544:26)
exited : 1

EmbeddedMan, 2015-11-03:
I have also found this exact same problem. I have "fixed" it by reverting to Node.js version 0.12.7 and serialport version 1.7.4.

I sure hope that this issue can be solved in the latest serialport code, as I'd much rather be using the latest Node.js versions.

livesankp, 2015-11-03:
Right. Not sure what is causing this. When i run serialport and jsdom, jquery separately it is working. But the moment i combine both of them it starts crashing. Reverting everything back to your suggestion was always working i tried out that before.

jacobrosenthal, 2015-11-09:
Can any of you still reproduce this with the newly released 2.0.4?

livesankp, 2015-11-11:
@jacobrosenthal i tried this.  Looks like this issue is fixed with node 4.2.2 and serialport 2.0.4
Will let you know if this happens again.

closed 2015-11-12 by jacobrosenthal

jonfaust, 2015-12-01:
The update seems to have fixed my problem too.  Thanks much!  The interesting thing was that it worked running on the Raspberry Pi (Raspbian), but crashed on my windows workstation.  There must be some differences in the implementation of something under the hood.