serialport/node-serialport #619: can not run under nodejs 5.0

ericdai, 2015-10-30:
anyone run it under nodejs5.0？

fivdi, 2015-11-02:
Yes, just tried serialport v2.0.2 with Node.js v5.0.0 on a Raspberry Pi 2 running Raspbian Jessie and everything installed and functioned as expected.

rk4an, 2015-11-14:
Hi,

Build style failed on my RPI B+ python 2 installed

10098 error Linux 4.1.12+
10099 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport"
10100 error node v5.0.0
10101 error npm  v3.4.1
10102 error code ELIFECYCLE
10103 error serialport@2.0.5 install: `node-pre-gyp install --fallback-to-build`
10103 error Exit status 1
10104 error Failed at the serialport@2.0.5 install script 'node-pre-gyp install --fallback-to-build'.
10104 error Make sure you have the latest version of node.js and npm installed.
10104 error If you do, this is most likely a problem with the serialport package,
10104 error not with npm itself.
10104 error Tell the author that this fails on your system:
10104 error     node-pre-gyp install --fallback-to-build
10104 error You can get their info via:
10104 error     npm owner ls serialport

fivdi, 2015-11-14:
Installing serialport v2.0.5 with Node.js v5.0.0 on a Raspberry Pi 2 running Raspbian Jessie worked for me:

```
pi@raspberrypi ~/dev $ node -v
v5.0.0

pi@raspberrypi ~/dev $ uname -a
Linux raspberrypi 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux

pi@raspberrypi ~/dev $ npm install serialport

> serialport@2.0.5 install /home/pi/dev/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/pi/dev/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
  COPY /home/pi/dev/node_modules/serialport/build/Release/node-v47-linux-arm/serialport.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/home/pi/dev/node_modules/serialport/build'
/home/pi/dev
└─┬ serialport@2.0.5 
  ├── async@0.9.0 
  ├── bindings@1.2.1 
  ├─┬ debug@2.2.0 
  │ └── ms@0.7.1 
  ├── nan@2.0.9 
  ├─┬ node-pre-gyp@0.6.13
  │ ├─┬ npmlog@1.2.1
  │ │ └─┬ are-we-there-yet@1.0.4
  │ │   └─┬ readable-stream@1.1.13
  │ │     ├── core-util-is@1.0.1 
  │ │     ├── inherits@2.0.1 
  │ │     ├── isarray@0.0.1 
  │ │     └── string_decoder@0.10.31 
  │ ├─┬ request@2.65.0 
  │ │ ├── aws-sign2@0.6.0 
  │ │ ├─┬ bl@1.0.0 
  │ │ │ └─┬ readable-stream@2.0.2
  │ │ │   ├── core-util-is@1.0.1 
  │ │ │   ├── inherits@2.0.1 
  │ │ │   ├── isarray@0.0.1 
  │ │ │   ├── process-nextick-args@1.0.3 
  │ │ │   ├── string_decoder@0.10.31 
  │ │ │   └── util-deprecate@1.0.2 
  │ │ ├── caseless@0.11.0 
  │ │ ├─┬ combined-stream@1.0.5 
  │ │ │ └── delayed-stream@1.0.0 
  │ │ ├── extend@3.0.0 
  │ │ ├── forever-agent@0.6.1 
  │ │ ├── form-data@1.0.0-rc3 
  │ │ ├─┬ har-validator@2.0.2 
  │ │ │ ├─┬ chalk@1.1.1 
  │ │ │ │ ├── ansi-styles@2.1.0 
  │ │ │ │ ├── escape-string-regexp@1.0.3 
  │ │ │ │ ├─┬ has-ansi@2.0.0 
  │ │ │ │ │ └── ansi-regex@2.0.0 
  │ │ │ │ ├─┬ strip-ansi@3.0.0 
  │ │ │ │ │ └── ansi-regex@2.0.0 
  │ │ │ │ └── supports-color@2.0.0 
  │ │ │ ├─┬ commander@2.9.0 
  │ │ │ │ └── graceful-readlink@1.0.1 
  │ │ │ ├─┬ is-my-json-valid@2.12.2
  │ │ │ │ ├── generate-function@2.0.0 
  │ │ │ │ ├─┬ generate-object-property@1.2.0 
  │ │ │ │ │ └── is-property@1.0.2 
  │ │ │ │ └── jsonpointer@2.0.0 
  │ │ │ └─┬ pinkie-promise@1.0.0 
  │ │ │   └── pinkie@1.0.0 
  │ │ ├─┬ hawk@3.1.0
  │ │ │ ├── cryptiles@2.0.5 
  │ │ │ ├── hoek@2.16.3 
  │ │ │ └── sntp@1.0.9 
  │ │ ├─┬ http-signature@0.11.0 
  │ │ │ ├── asn1@0.1.11 
  │ │ │ ├── assert-plus@0.1.5 
  │ │ │ └── ctype@0.5.3 
  │ │ ├── isstream@0.1.2 
  │ │ ├── json-stringify-safe@5.0.1 
  │ │ ├─┬ mime-types@2.1.7 
  │ │ │ └── mime-db@1.19.0 
  │ │ ├── oauth-sign@0.8.0 
  │ │ ├── qs@5.2.0 
  │ │ └── tunnel-agent@0.4.1 
  │ ├─┬ rimraf@2.4.3
  │ │ └─┬ glob@5.0.15
  │ │   └── inherits@2.0.1 
  │ ├─┬ tar@2.2.1
  │ │ └── inherits@2.0.1 
  │ └─┬ tar-pack@2.0.1
  │   ├─┬ fstream@0.1.31
  │   │ └── inherits@2.0.1 
  │   ├─┬ fstream-ignore@0.0.7
  │   │ └── inherits@2.0.1 
  │   ├─┬ readable-stream@1.0.33
  │   │ ├── core-util-is@1.0.1 
  │   │ ├── inherits@2.0.1 
  │   │ ├── isarray@0.0.1 
  │   │ └── string_decoder@0.10.31 
  │   └─┬ tar@0.1.20
  │     └── inherits@2.0.1 
  ├─┬ node-pre-gyp-github@1.0.5  (git://github.com/jacobrosenthal/node-pre-gyp-github.git#cd28d5e71cd99b1d64d786a16a03db811d757234)
  │ ├─┬ commander@2.9.0 
  │ │ └── graceful-readlink@1.0.1 
  │ ├── github@0.2.4 
  │ ├── mime@1.3.4 
  │ └─┬ request@2.65.0 
  │   ├── aws-sign2@0.6.0 
  │   ├─┬ bl@1.0.0 
  │   │ └─┬ readable-stream@2.0.4 
  │   │   ├── core-util-is@1.0.1 
  │   │   ├── inherits@2.0.1 
  │   │   ├── isarray@0.0.1 
  │   │   ├── process-nextick-args@1.0.3 
  │   │   ├── string_decoder@0.10.31 
  │   │   └── util-deprecate@1.0.2 
  │   ├── caseless@0.11.0 
  │   ├─┬ combined-stream@1.0.5 
  │   │ └── delayed-stream@1.0.0 
  │   ├── extend@3.0.0 
  │   ├── forever-agent@0.6.1 
  │   ├─┬ form-data@1.0.0-rc3 
  │   │ └── async@1.5.0 
  │   ├─┬ har-validator@2.0.2 
  │   │ ├─┬ chalk@1.1.1 
  │   │ │ ├── ansi-styles@2.1.0 
  │   │ │ ├── escape-string-regexp@1.0.3 
  │   │ │ ├─┬ has-ansi@2.0.0 
  │   │ │ │ └── ansi-regex@2.0.0 
  │   │ │ ├── strip-ansi@3.0.0 
  │   │ │ └── supports-color@2.0.0 
  │   │ ├─┬ is-my-json-valid@2.12.3 
  │   │ │ ├── generate-function@2.0.0 
  │   │ │ ├─┬ generate-object-property@1.2.0 
  │   │ │ │ └── is-property@1.0.2 
  │   │ │ ├── jsonpointer@2.0.0 
  │   │ │ └── xtend@4.0.1 
  │   │ └─┬ pinkie-promise@1.0.0 
  │   │   └── pinkie@1.0.0 
  │   ├─┬ hawk@3.1.1 
  │   │ ├── boom@2.10.1 
  │   │ ├── cryptiles@2.0.5 
  │   │ ├── hoek@2.16.3 
  │   │ └── sntp@1.0.9 
  │   ├─┬ http-signature@0.11.0 
  │   │ ├── asn1@0.1.11 
  │   │ ├── assert-plus@0.1.5 
  │   │ └── ctype@0.5.3 
  │   ├── isstream@0.1.2 
  │   ├── json-stringify-safe@5.0.1 
  │   ├─┬ mime-types@2.1.7 
  │   │ └── mime-db@1.19.0 
  │   ├── node-uuid@1.4.7 
  │   ├── oauth-sign@0.8.0 
  │   ├── qs@5.2.0 
  │   ├── stringstream@0.0.5 
  │   ├── tough-cookie@2.2.1 
  │   └── tunnel-agent@0.4.1 
  ├─┬ optimist@0.6.1 
  │ ├── minimist@0.0.10 
  │ └── wordwrap@0.0.3 
  └── sf@0.1.7 

npm WARN ENOENT ENOENT: no such file or directory, open '/home/pi/dev/package.json'
npm WARN EPACKAGEJSON /home/pi/dev No description
npm WARN EPACKAGEJSON /home/pi/dev No repository field.
npm WARN EPACKAGEJSON /home/pi/dev No README data
npm WARN EPACKAGEJSON /home/pi/dev No license field.
pi@raspberrypi ~/dev $ 

```

rk4an, 2015-11-15:
Thanks,

Here the ouput of the install:

dj-hedgehog, 2015-11-25:
Doesn't compile on OS X `10.11.2` and node `5.1.0`

```
npm i serialport

> serialport@2.0.5 install /Users/my_user/my_project/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp: /Users/my_user/.node-gyp/5.1.0/common.gypi not found (cwd: /Users/my_user/my_project/node_modules/serialport) while reading includes of binding.gyp while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/Users/my_user/my_project/node_modules/node-gyp/lib/configure.js:357:16)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 15.0.0
gyp ERR! command "/Users/my_user/.nvm/versions/node/v5.1.0/bin/node" "/Users/my_user/my_project/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/Users/my_user/my_project/node_modules/serialport/build/Release/node-v47-darwin-x64/serialport.node" "--module_name=serialport" "--module_path=/Users/my_user/my_project/node_modules/serialport/build/Release/node-v47-darwin-x64"
gyp ERR! cwd /Users/my_user/my_project/node_modules/serialport
gyp ERR! node -v v5.1.0
gyp ERR! node-gyp -v v2.0.2
gyp ERR! not ok
```

deadprogram, 2015-11-26:
Our Travis build for https://github.com/orbotix/sphero.js fails on Node 5.0 but passes for version 0.10 - 4.2.2 using `node-serialport` v2.0.5

https://travis-ci.org/orbotix/sphero.js/jobs/93281788

```
> serialport@2.0.5 install /home/travis/build/orbotix/sphero.js/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory `/home/travis/build/orbotix/sphero.js/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from /home/travis/.node-gyp/5.0.0/include/node/node.h:42:0,
                 from ../../nan/nan.h:24,
                 from ../src/serialport.h:5,
                 from ../src/serialport.cpp:3:
/home/travis/.node-gyp/5.0.0/include/node/v8.h:336:1: error: expected unqualified-id before ‘using’
/home/travis/.node-gyp/5.0.0/include/node/v8.h:469:1: error: expected unqualified-id before ‘using’
/home/travis/.node-gyp/5.0.0/include/node/v8.h:856:1: error: expected unqualified-id before ‘using’
In file included from ../../nan/nan.h:182:0,
                 from ../src/serialport.h:5,
                 from ../src/serialport.cpp:3:
...
```

fivdi, 2015-11-26:
@deadprogram it looks like the Travis build for Node.js v5.0 is using a version of g++ that is too old. To compile native add-ons like serialport with Node.js v5.0, g++ v4.8 or higher is required.

deadprogram, 2015-11-26:
That information should probably be part of the README, and/or the long-languishing wiki. Also, the Travis build should probably be updated to test these versions. I could probably do this right now, @jacobrosenthal any interest in a PR adding these things?

deadprogram, 2015-12-04:
Hello, all. Just submitted a PR that should handle the Travis/AppVeyor builds for Node.js 5.1. Please let me know what you think.

deadprogram, 2015-12-04:
This is the PR: https://github.com/voodootikigod/node-serialport/pull/640

closed 2016-03-28 by reconbot