serialport/node-serialport #620: can not install ...

Apollon77, 2015-11-04:
I try to install it on my CubieTruck and I get this:

 npm install -g serialport
/usr/local/bin/serialportlist -> /usr/local/lib/node_modules/serialport/bin/serialportList.js
/usr/local/bin/serialportterm -> /usr/local/lib/node_modules/serialport/bin/serialportTerminal.js

> serialport@2.0.2 install /usr/local/lib/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.12.6"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/serialport/.node-gyp"
make: Entering directory `/usr/local/lib/node_modules/serialport/build'
make: *** No rule to make target `../.node-gyp/0.12.6/common.gypi', needed by `Makefile'.  Stop.
make: Leaving directory `/usr/local/lib/node_modules/serialport/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.4.109-sunxi
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/local/lib/node_modules/serialport/build/serialport/v2.0.2/Release/node-v14-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/usr/local/lib/node_modules/serialport/build/serialport/v2.0.2/Release/node-v14-linux-arm"
gyp ERR! cwd /usr/local/lib/node_modules/serialport
gyp ERR! node -v v0.12.6
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/serialport/build/serialport/v2.0.2/Release/node-v14-linux-arm/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/serialport/build/serialport/v2.0.2/Release/node-v14-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:1015:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1087:5)
node-pre-gyp ERR! System Linux 3.4.109-sunxi
node-pre-gyp ERR! command "node" "/usr/local/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/local/lib/node_modules/serialport
node-pre-gyp ERR! node -v v0.12.6
node-pre-gyp ERR! node-pre-gyp -v v0.6.13
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/serialport/build/serialport/v2.0.2/Release/node-v14-linux-arm/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/serialport/build/serialport/v2.0.2/Release/node-v14-linux-arm' (1)
npm ERR! Linux 3.4.109-sunxi
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "serialport"
npm ERR! node v0.12.6
npm ERR! npm  v2.11.2
npm ERR! code ELIFECYCLE

npm ERR! serialport@2.0.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@2.0.2 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /opt/homebridge/npm-debug.log

Any idea?!

Apollon77, 2015-11-05:
found parameter "--unsafe-perm" may work ... trying it

jacobrosenthal, 2015-11-09:
That do anything?

You're falling back to building source for some reason. I thought binaries were good for 2.0.2 but you never know. Try 2.0.4 I know those exist. 

I find npm install loses or obscures logs, try a `node-pre-gyp install --fallback-to-build` directly to see what that output is, too

closed 2015-11-22 by jacobrosenthal