serialport/node-serialport #622: On raspberry Pi list function returns undefined when only  /dev/ttyAMA0 is available

johnofleek, 2015-11-05:
Hi

Using standard code
var serialPort = require("serialport");
serialPort.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(port.comName);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});

and only /dev/ttyAMA0 available - I get 
 ports.forEach(function(port) {
       ^
TypeError: Cannot read property 'forEach' of undefined
    at /home/pi/hlhat/sp.js:3:8
    at /home/pi/hlhat/node_modules/serialport/serialport.js:565:11
    at FSReqWrap.oncomplete (fs.js:95:15)

If I plug in a USB serial port I get - (also ok with /dev/ttyACMx)
/dev/ttyUSB0
usb-FTDI_US232R_FTD6BP91-if00-port0
FTDI

fivdi, 2015-11-07:
Yes, `list` doesn't detect `/dev/ttyAMA0`. However, it's still possible to use `/dev/ttyAMA0`
```
new SerialPort('/dev/ttyAMA0', {baudRate: 115200})
```

jacobrosenthal, 2015-11-09:
It should hopefully detect ttyAMA0 as of 2.0.3, Id love some feedback on that, meet me over at https://github.com/voodootikigod/node-serialport/issues/610

closed 2015-11-09 by jacobrosenthal