serialport/node-serialport #624: Failing install on BeagleBone Black

jandrieu, 2015-11-07:
I'm trying and failing to get serialport to work on BeagleBone Black.

I could build node from source and npm mostly seems to work, but serialport can't install.

The verbose output from the install follows. GIthub wouldn't let me attach the npm-log file.

-j

```
ubuntu@arm:~/.ssh/Amigos/mote/software/blinkie/liason$ npm install serialport --verbose
npm info it worked if it ends with ok
npm verb cli [ '/usr/local/bin/node',
npm verb cli   '/usr/local/bin/npm',
npm verb cli   'install',
npm verb cli   'serialport',
npm verb cli   '--verbose' ]
npm info using npm@2.14.7
npm info using node@v4.2.2
npm verb install initial load of /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/package.json
npm verb readDependencies loading dependencies from /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/package.json
npm verb cache add spec serialport@^1.7.4
npm verb addNamed ">=1.7.4 <2.0.0" is a valid semver range for serialport
npm verb addNameRange registry:https://registry.npmjs.org/serialport not in flight; fetching
npm verb request uri https://registry.npmjs.org/serialport
npm verb request no auth needed
npm info attempt registry request try #1 at 22:02:30
npm verb request id f99525ca142708ed
npm verb etag "3TN0FG3JGEJR7FRUMM4IGM6ZL"
npm http request GET https://registry.npmjs.org/serialport
npm http 304 https://registry.npmjs.org/serialport
npm verb headers { date: 'Sat, 07 Nov 2015 22:02:29 GMT',
npm verb headers   via: '1.1 varnish',
npm verb headers   'cache-control': 'max-age=60',
npm verb headers   etag: '"3TN0FG3JGEJR7FRUMM4IGM6ZL"',
npm verb headers   age: '0',
npm verb headers   connection: 'keep-alive',
npm verb headers   'x-served-by': 'cache-lax1428-LAX',
npm verb headers   'x-cache': 'HIT',
npm verb headers   'x-cache-hits': '1',
npm verb headers   'x-timer': 'S1446933749.381191,VS0,VE47',
npm verb headers   vary: 'Accept' }
npm verb etag https://registry.npmjs.org/serialport from cache
npm verb get saving serialport to /home/ubuntu/.npm/registry.npmjs.org/serialport/.cache.json
npm verb addNamed "1.7.4" is a plain semver version for serialport
npm verb afterAdd /home/ubuntu/.npm/serialport/1.7.4/package/package.json not in flight; writing
npm verb afterAdd /home/ubuntu/.npm/serialport/1.7.4/package/package.json written
npm info install serialport@1.7.4 into /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason
npm info installOne serialport@1.7.4
npm verb installOne of serialport to /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason not in flight; installing
npm verb lock using /home/ubuntu/.npm/_locks/serialport-f68ddc7d01f7a2f1.lock for /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/serialport
npm verb unbuild node_modules/serialport
npm verb gentlyRm don't care about contents; nuking /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/serialport
npm verb tar unpack /home/ubuntu/.npm/serialport/1.7.4/package.tgz
npm verb tar unpacking to /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/serialport
npm verb gentlyRm don't care about contents; nuking /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/serialport
npm verb write writing to /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/serialport/package.json
npm info preinstall serialport@1.7.4
npm verb readDependencies loading dependencies from /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/serialport/package.json
npm verb readDependencies loading dependencies from /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/serialport/package.json
npm verb targetResolver reading package data from /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/serialport/node_modules/node-pre-gyp/package.json
npm info package.json node-pre-gyp@0.6.7 license should be a valid SPDX license expression
npm verb cache add spec async@0.9.0
npm verb addNamed "0.9.0" is a plain semver version for async
npm verb addNameVersion registry:https://registry.npmjs.org/async not in flight; fetching
npm verb cache add spec bindings@1.2.1
npm verb addNamed "1.2.1" is a plain semver version for bindings
npm verb addNameVersion registry:https://registry.npmjs.org/bindings not in flight; fetching
npm verb cache add spec debug@^2.1.1
npm verb addNamed ">=2.1.1 <3.0.0" is a valid semver range for debug
npm verb addNameRange registry:https://registry.npmjs.org/debug not in flight; fetching
[... 784 lines cut ...]
npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.
npm verb exit [ 1, true ]
npm verb unbuild node_modules/serialport
npm info preuninstall serialport@1.7.4
npm info uninstall serialport@1.7.4
npm verb unbuild rmStuff serialport@1.7.4 from /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules
npm verb gentlyRm vacuuming from /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/.bin/serialportlist up to /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason
npm verb gentlyRm vacuuming from /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/.bin/serialportterm up to /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason
npm info postuninstall serialport@1.7.4
npm verb gentlyRm don't care about contents; nuking /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/node_modules/serialport

npm ERR! Please include the following file with any support request:
npm ERR!     /home/ubuntu/.ssh/Amigos/mote/software/blinkie/liason/npm-debug.log
ubuntu@arm:~/.ssh/Amigos/mote/software/blinkie/liason$ 
```

jandrieu, 2015-11-08:
Good news.

Reverting to node v0.10.40 worked.

elafargue, 2015-11-08:
You need the 2.X.Y series of node-serialport on the new versions of Node...

jacobrosenthal, 2015-11-09:
Thats correct.

closed 2015-11-09 by jacobrosenthal