serialport/node-serialport #625: Error: Command failed: git -c core.longpaths=true config --get remote.origin.url

jcarpe, 2015-11-10:
I've tried using npm install, but it seems as though the title command is failing.

```
npm WARN addRemoteGit Error: Command failed: git -c core.longpaths=true config --get remote.origin.url
npm WARN addRemoteGit
npm WARN addRemoteGit     at ChildProcess.exithandler (child_process.js:203:12)
npm WARN addRemoteGit     at emitTwo (events.js:87:13)
npm WARN addRemoteGit     at ChildProcess.emit (events.js:172:7)
npm WARN addRemoteGit     at maybeClose (internal/child_process.js:818:16)
npm WARN addRemoteGit     at Socket.<anonymous> (internal/child_process.js:319:11)
npm WARN addRemoteGit     at emitOne (events.js:77:13)
npm WARN addRemoteGit     at Socket.emit (events.js:169:7)
npm WARN addRemoteGit     at Pipe._onclose (net.js:469:12)
npm WARN addRemoteGit  jacobrosenthal/node-pre-gyp-github#production resetting remote C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-00f1b2658a8ab0261b1b6e8aa994ae37 because of error: { [Error: Command failed: git -c core.longpaths=true config --get remote.origin.url
npm WARN addRemoteGit ]
npm WARN addRemoteGit   killed: false,
npm WARN addRemoteGit   code: 1,
npm WARN addRemoteGit   signal: null,
npm WARN addRemoteGit   cmd: 'git -c core.longpaths=true config --get remote.origin.url' }
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git://github.com/jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-00f1b2658a8ab0261b1b6e8aa994ae37: Cloning into bare repository 'C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-00f1b2658a8ab0261b1b6e8aa994ae37'...
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git://github.com/jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-00f1b2658a8ab0261b1b6e8aa994ae37: fatal: unable to connect to github.com:
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git://github.com/jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-00f1b2658a8ab0261b1b6e8aa994ae37: github.com[0: 192.30.252.129]: errno=No such file or directory
npm WARN addRemoteGit Error: Command failed: git -c core.longpaths=true config --get remote.origin.url
npm WARN addRemoteGit
npm WARN addRemoteGit     at ChildProcess.exithandler (child_process.js:203:12)
npm WARN addRemoteGit     at emitTwo (events.js:87:13)
npm WARN addRemoteGit     at ChildProcess.emit (events.js:172:7)
npm WARN addRemoteGit     at maybeClose (internal/child_process.js:818:16)
npm WARN addRemoteGit     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
npm WARN addRemoteGit  jacobrosenthal/node-pre-gyp-github#production resetting remote C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-https-github-com-jacobrosenthal-node-pre-gyp-github-git-production-1e8dd7eab6415a0b31fc9e88e23213e9 because of error: { [Error: Command failed: git -c core.longpaths=true config --get remote.origin.url
npm WARN addRemoteGit ]
npm WARN addRemoteGit   killed: false,
npm WARN addRemoteGit   code: 1,
npm WARN addRemoteGit   signal: null,
npm WARN addRemoteGit   cmd: 'git -c core.longpaths=true config --get remote.origin.url' }
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror https://github.com/jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-https-github-com-jacobrosenthal-node-pre-gyp-github-git-production-1e8dd7eab6415a0b31fc9e88e23213e9: Cloning into bare repository 'C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-https-github-com-jacobrosenthal-node-pre-gyp-github-git-production-1e8dd7eab6415a0b31fc9e88e23213e9'...
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror https://github.com/jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-https-github-com-jacobrosenthal-node-pre-gyp-github-git-production-1e8dd7eab6415a0b31fc9e88e23213e9: fatal: Unable to find remote helper for 'https'
npm WARN addRemoteGit Error: Command failed: git -c core.longpaths=true config --get remote.origin.url
npm WARN addRemoteGit
npm WARN addRemoteGit     at ChildProcess.exithandler (child_process.js:203:12)
npm WARN addRemoteGit     at emitTwo (events.js:87:13)
npm WARN addRemoteGit     at ChildProcess.emit (events.js:172:7)
npm WARN addRemoteGit     at maybeClose (internal/child_process.js:818:16)
npm WARN addRemoteGit     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
npm WARN addRemoteGit  jacobrosenthal/node-pre-gyp-github#production resetting remote C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390 because of error: { [Error: Command failed: git -c core.longpaths=true config --get remote.origin.url
npm WARN addRemoteGit ]
npm WARN addRemoteGit   killed: false,
npm WARN addRemoteGit   code: 1,
npm WARN addRemoteGit   signal: null,
npm WARN addRemoteGit   cmd: 'git -c core.longpaths=true config --get remote.origin.url' }
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git@github.com:jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390: Cloning into bare repository 'C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390'...
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git@github.com:jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390: Permission denied (publickey).
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git@github.com:jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390: fatal: Could not read from remote repository.
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git@github.com:jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390:
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git@github.com:jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390: Please make sure you have the correct access rights
npm ERR! git clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git@github.com:jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390: and the repository exists.
npm WARN install Couldn't install optional dependency: Command failed: git -c core.longpaths=true clone --template=C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git@github.com:jacobrosenthal/node-pre-gyp-github.git C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390
npm WARN install Cloning into bare repository 'C:\Users\SapientBoston\AppData\Roaming\npm-cache\_git-remotes\git-github-com-jacobrosenthal-node-pre-gyp-github-git-production-d32d5d69965585ea64fc38a162bde390'...
npm WARN install Permission denied (publickey).
npm WARN install fatal: Could not read from remote repository.
npm WARN install
npm WARN install Please make sure you have the correct access rights
npm WARN install and the repository exists.
npm WARN install
```

It seems that this *might* be an issue with git on my own machine, but I am not sure what that would be.

jcarpe, 2015-11-10:
FYI, I also tried changing the protocol for git globally to https, but the error is exactly the same (thought it might be a company security policy thing on my end).

jcarpe, 2015-11-10:
So, it looks like my issue is 2 fold; there was some ignorance around my use of git on Windows (first time for everything) as well as a Visual Studio configuration in npm. 

To resolve the MSBUILD issue, I had to configure npm for Visual Studio 2015.

```
npm config set msvs_version 2015 --global
```

I had to do this because 2013 Desktop version is no longer available (as noted in the serial port documentation).

closed 2015-11-10 by jcarpe

consideRatio, 2015-11-27:
Can you explain a bit about the "ignorance around my use of git on windows" ?
(I have the same issue)