serialport/node-serialport #626: DEVLINKS.split problem with Node v4.1.2 and v5

ysangkok, 2015-11-10:
I get this issue:

```
/tmp/node-v5.0.0-linux-x64/bin $ ./particle identify
/tmp/node-v5.0.0-linux-x64/lib/node_modules/particle-cli/node_modules/serialport/serialport.js:541
      var pnpId = as_json.DEVLINKS.split(' ')[0];
                                  ^

TypeError: Cannot read property 'split' of undefined
    at udev_parser (/tmp/node-v5.0.0-linux-x64/lib/node_modules/particle-cli/node_modules/serialport/serialport.js:541:35)
    at /tmp/node-v5.0.0-linux-x64/lib/node_modules/particle-cli/node_modules/serialport/serialport.js:592:11
    at ChildProcess.exithandler (child_process.js:194:7)
    at emitTwo (events.js:87:13)
    at ChildProcess.emit (events.js:172:7)
    at maybeClose (internal/child_process.js:818:16)
    at Socket.<anonymous> (internal/child_process.js:319:11)
    at emitOne (events.js:77:13)
    at Socket.emit (events.js:169:7)
    at Pipe._onclose (net.js:469:12)
```

I can provide more details if you'd like.

kzvaigzne, 2015-11-11:
I'm getting the same issue with Node v4.2.2 and Linux Mint 17.

So far it only occurs when trying to list serial devices

ysangkok, 2015-11-11:
I fixed my issue by deleting `/usr/local/lib/node_modules` and symlinking it to `/usr/lib/node_modules`.

ysangkok, 2015-11-11:
No, sorry, that solved a different issue. But I worked around this by commenting out the lines relating to `pnpId`.

jacobrosenthal, 2015-11-12:
merged a fix, will go out as 2.0.5

closed 2015-11-12 by jacobrosenthal