serialport/node-serialport #628: List ports on Windows 10/8.1

Marabyte, 2015-11-12:
I've been running in a couple of issues to get serialport running on Windows 10/8.1

First it failed to install throwing error: `MSBUILD : error MSB4132: The tools version "2.0" is unrecognized"`
I installed Visual Studio Community 2015 with the Common Tools for Visual C++ 2015 and then 
```
$ npm install serialport --save --msvs_version=2015 
```
Only to get `error MSB8020: The build tools for v140 (Platform Toolset = 'v140') cannot be found.`

Apparently the compiler was looking in the wrong folder, so I needed to set it to 
```
set VCTargetsPath=C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140
``` 
It throws a couple of warnings but builds successfully.

On my node app, I'm trying to list the serial ports using the example on the serialport readme (https://github.com/voodootikigod/node-serialport#list-ports) but it just return an empty array.
I have an USB printer connected and my mobile by bluetooth but nothing is listed.

I've tested with both Windows 10 and 8.1, node 4.2.2 and node 5.0 and serialport 2.0.4.

Any ideas? Could it be the VS2015 C++ build? I'm trying to compile using VS2013 (V120) but can't find the build tools for it.

Cheers,
Hugo

jacobrosenthal, 2015-11-12:
2.0.4 has hosted binaries for windows. Can you show your npm install log so we can see why the binary download fails?

Make sure some other program can find your com ports and that theyre listed in Device manager.

Marabyte, 2015-11-12:
> serialport@2.0.4 install C:\Users\hnogu\Documents\Sites\node-printer-zebra\nod
e_modules\serialport
> node-pre-gyp install --fallback-to-build

C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\serialport>if not
 defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin
\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" clean )  else (node
 clean )

C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\serialport>if not
 defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin
\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" configure --fallback
-to-build --module=C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_module
s\serialport\build\Release\node-v46-win32-x64\serialport.node --module_name=seri
alport --module_path=C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modu
les\serialport\build\Release\node-v46-win32-x64 --msvs_version=2015 )  else (nod
e  configure --fallback-to-build --module=C:\Users\hnogu\Documents\Sites\node-pr
inter-zebra\node_modules\serialport\build\Release\node-v46-win32-x64\serialport.
node --module_name=serialport --module_path=C:\Users\hnogu\Documents\Sites\node-
printer-zebra\node_modules\serialport\build\Release\node-v46-win32-x64 --msvs_ve
rsion=2015 )

C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\serialport>if not
 defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin
\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" build --fallback-to-
build --module=C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\se
rialport\build\Release\node-v46-win32-x64\serialport.node --module_name=serialpo
rt --module_path=C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\
serialport\build\Release\node-v46-win32-x64 )  else (node  build --fallback-to-b
uild --module=C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\ser
ialport\build\Release\node-v46-win32-x64\serialport.node --module_name=serialpor
t --module_path=C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\s
erialport\build\Release\node-v46-win32-x64 )
Building the projects in this solution one at a time. To enable parallel build,
please add the "/m" switch.
  serialport.cpp
  serialport_win.cpp
  enumser.cpp
..\src\win\enumser.cpp(443): warning C4996: 'GetVersionExA': was declared depre
cated [C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\serialpor
t\build\serialport.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433): note: s
  ee declaration of 'GetVersionExA'
c:\users\hnogu\documents\sites\node-printer-zebra\node_modules\serialport\src\w
in\stdstring.h(2123): warning C4311: 'reinterpret_cast': pointer truncation fro
m 'const void *' to 'unsigned long' (compiling source file ..\src\win\enumser.c
pp) [C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\serialport\
build\serialport.vcxproj]
  c:\users\hnogu\documents\sites\node-printer-zebra\node_modules\serialport\src
  \win\stdstring.h(2117): note: while compiling class template member function
  'bool CStdStr<wchar_t>::TryLoad(const void *)' (compiling source file ..\src\
  win\enumser.cpp)
  c:\users\hnogu\documents\sites\node-printer-zebra\node_modules\serialport\src
  \win\stdstring.h(2171): note: see reference to function template instantiatio
  n 'bool CStdStr<wchar_t>::TryLoad(const void *)' being compiled (compiling so
  urce file ..\src\win\enumser.cpp)
  c:\users\hnogu\documents\sites\node-printer-zebra\node_modules\serialport\src
  \win\stdstring.h(3982): note: see reference to class template instantiation '
  CStdStr<wchar_t>' being compiled (compiling source file ..\src\win\enumser.cp
  p)
c:\users\hnogu\documents\sites\node-printer-zebra\node_modules\serialport\src\w
in\stdstring.h(2123): warning C4302: 'reinterpret_cast': truncation from 'const
 void *' to 'unsigned long' (compiling source file ..\src\win\enumser.cpp) [C:\
Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\serialport\build\se
rialport.vcxproj]
c:\users\hnogu\documents\sites\node-printer-zebra\node_modules\serialport\src\w
in\stdstring.h(2126): warning C4311: 'reinterpret_cast': pointer truncation fro
m 'const void *' to 'unsigned long' (compiling source file ..\src\win\enumser.c
pp) [C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\serialport\
build\serialport.vcxproj]
[... 112 lines cut ...]
T' to 'LPVOID' of greater size [C:\Users\hnogu\Documents\Sites\node-printer-zeb
ra\node_modules\serialport\build\serialport.vcxproj]
..\src\win\disphelper.c(1164): warning C4267: '=': conversion from 'size_t' to
'UINT', possible loss of data [C:\Users\hnogu\Documents\Sites\node-printer-zebr
a\node_modules\serialport\build\serialport.vcxproj]
     Creating library C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_mo
  dules\serialport\build\Release\serialport.lib and object C:\Users\hnogu\Docum
  ents\Sites\node-printer-zebra\node_modules\serialport\build\Release\serialpor
  t.exp
  Generating code
  Finished generating code
  serialport.vcxproj -> C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_
  modules\serialport\build\Release\\serialport.node
  Copying C:\Users\hnogu\Documents\Sites\node-printer-zebra\node_modules\serial
  port\build\Release\/serialport.node to C:/Users/hnogu/Documents/Sites/node-pr
  inter-zebra/node_modules/serialport/build/Release/node-v46-win32-x64\serialpo
  rt.node
          1 file(s) copied.
serialport@2.0.4 node_modules\serialport
├── bindings@1.2.1
├── async@0.9.0
├── sf@0.1.7
├── debug@2.2.0 (ms@0.7.1)
├── nan@2.0.9
└── optimist@0.6.1 (wordwrap@0.0.3, minimist@0.0.10)

jacobrosenthal, 2015-11-12:
That seems incomplete to me, its missing the success or failure of node-pre gyp downloading the binary.

Heres my successful log
```
Heres what that looks like

C:\Users\jacobrosenthal\Desktop\testserialport>npm i serialport@2.0.5
npm WARN package.json testserialport@1.0.0 No description
npm WARN package.json testserialport@1.0.0 No repository field.
npm WARN package.json testserialport@1.0.0 No README data
/
> serialport@2.0.5 install C:\Users\jacobrosenthal\Desktop\testserialport\node_m
odules\serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "C:\Users\jacobrosenthal\Desktop\testserialport\node_modul
es\serialport\build\Release\node-v46-win32-ia32\serialport.node" is installed vi
a remote
serialport@2.0.5 node_modules\serialport
├── bindings@1.2.1
├── sf@0.1.7
├── async@0.9.0
├── nan@2.0.9
├── debug@2.2.0 (ms@0.7.1)
└── optimist@0.6.1 (wordwrap@0.0.3, minimist@0.0.10)

C:\Users\jacobrosenthal\Desktop\testserialport>
```

Marabyte, 2015-11-12:
I've deleted the node_modules folder and run `npm cache clear`

After that I tried to install again and got the `[serialport] : Success`
It installed version 2.0.5 instead of 2.0.4, but I cant still list any devices.

I've double checked and they are correctly being displayed on Device Manager

reconbot, 2016-03-31:
Hi @Marabyte I'm happy to help debug if the latest version (or beta) is still giving you issues. Please reopen if that's the case. Sorry we let this drop!

closed 2016-03-31 by reconbot