serialport/node-serialport #634: On Linux serialport.list produces different results

sarakusha, 2015-11-25:
pnpId can be either /dev/serial/by-path or /dev/serial/by-id at different time.
This is due to the fact that the output of the command `udevadm info` is different
```bash
DEVLINKS=/dev/serial/by-id/pci-NATA_Siolynx2_C8T6VI1F-if00-port0 /dev/serial/by-path/pci-0000:00:14.0-usb-0:2:1.0-port0
```
or
```bash
DEVLINKS=/dev/serial/by-path/pci-0000:00:14.0-usb-0:2:1.0-port0 /dev/serial/by-id/pci-NATA_Siolynx2_C8T6VI1F-if00-port0
```
The program always selects the first item:
```js
var pnpId;
if(as_json.DEVLINKS) {
        pnpId = as_json.DEVLINKS.split(' ')[0];
        pnpId = pnpId.substring(pnpId.lastIndexOf('/') + 1);
}
```

jacobrosenthal, 2015-11-25:
We just removed /dev/serial lookups as theyre not available on all platforms. See https://github.com/voodootikigod/node-serialport/issues/610

sarakusha, 2015-11-25:
I left out this option (My English is poor)
```js
 pnpId = matches && matches[1] || as_json.DEVLINKS.split(' ')[0];
```
if matches === null then pnpId=as_json.DEVLINKS.split(' ')[0];
If there are two options (by-path and by-id) , you can not choose the first, because order is not defined
The result should be the same and do not change when repeated call

sarakusha, 2015-11-25:
Ubuntu 15.10
```js
var sp = require('serialport');

setInterval(function() {
     sp.list(function(err, ports) {
         ports.forEach(port=>port.PnpId && console.log(port.pnpId));
     });
}, 1000);
```
output
```bash
pci-NATA_Siolynx2_C8T6VI1F-if00-port0
pci-NATA_Siolynx2_C8T6VI1F-if00-port0
pci-0000:00:14.0-usb-0:2:1.0-port0
pci-0000:00:14.0-usb-0:2:1.0-port0
pci-NATA_Siolynx2_C8T6VI1F-if00-port0
pci-NATA_Siolynx2_C8T6VI1F-if00-port0
pci-NATA_Siolynx2_C8T6VI1F-if00-port0
pci-0000:00:14.0-usb-0:2:1.0-port0
pci-NATA_Siolynx2_C8T6VI1F-if00-port0
pci-NATA_Siolynx2_C8T6VI1F-if00-port0
pci-0000:00:14.0-usb-0:2:1.0-port0
pci-NATA_Siolynx2_C8T6VI1F-if00-port0
pci-0000:00:14.0-usb-0:2:1.0-port0
^C

```

sarakusha, 2015-11-25:
In the example above there is only one port, and it is displayed in different ways

reconbot, 2016-03-31:
It should only show one way now. Fixed in master and will be in the next beta. =) Please it out and let us know if it works for you!

closed 2016-03-31 by reconbot