serialport/node-serialport #636: Installation failed - Mac OS X El Capitan 10.11.1

was-clement, 2015-11-26:
Hi there,

I'm trying to work with the `cylon-neurosky`module but it requires your package and I cannot install node-serialport using `sudo npm i`.

System info:
`node --version` v5.1.0
`npm --version` 3.3.12
`python --version` Python 2.7.10
`gcc --version`
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.0.0 (clang-700.1.76)
Target: x86_64-apple-darwin15.0.0
Thread model: posix

Output:
```
sudo npm i

> serialport@1.7.4 install /Users/clementprocureur/node_project/soutenance/node_modules/serialport
> node-pre-gyp install --fallback-to-build 

  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../../nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../../nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../../nan/nan.h:661:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../../nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/clementprocureur/.node-gyp/5.1.0/include/node/node_buffer.h:31:40: note: 
      candidate function not viable: no known conversion from 'uint32_t'
      (aka 'unsigned int') to 'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/clementprocureur/.node-gyp/5.1.0/include/node/node_buffer.h:43:40: note: 
      candidate function not viable: 2nd argument ('const char *') would lose
      const qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/clementprocureur/.node-gyp/5.1.0/include/node/node_buffer.h:28:40: note: 
      candidate function not viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/clementprocureur/.node-gyp/5.1.0/include/node/node_buffer.h:36:40: note: 
      candidate function not viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../../nan/nan.h:676:12: error: no viable conversion from
      'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/clementprocureur/.node-gyp/5.1.0/include/node/v8.h:210:7: note: 
      candidate constructor (the implicit copy constructor) not viable: no known
      conversion from 'v8::MaybeLocal<v8::Object>' to 'const
      v8::Local<v8::Object> &' for 1st argument
class Local {
      ^
/Users/clementprocureur/.node-gyp/5.1.0/include/node/v8.h:210:7: note: 
      candidate constructor (the implicit move constructor) not viable: no known
      conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&'
      for 1st argument
[... 121 lines cut ...]
npm ERR! Darwin 15.0.0
npm ERR! argv "/usr/local/Cellar/node/5.1.0/bin/node" "/usr/local/bin/npm" "i"
npm ERR! node v5.1.0
npm ERR! npm  v3.3.12
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/clementprocureur/node_project/soutenance/npm-debug.log
```
I try severals tricks before posting the issue but anything worked. Please tell me if you need the `npm-debug.log`file.

Thanks a lot !

rwaldron, 2015-11-26:
Node.js 5.x is not yet supported.

was-clement, 2015-11-26:
@rwaldron Ok thanks, so on witch version I have to downgrad ?

was-clement, 2015-11-26:
After several tests 0.12 is good.

closed 2015-11-26 by was-clement

rwaldron, 2015-11-26:
I would've suggested 4.2.x

brunocascio, 2016-01-06:
@rwaldron +1