serialport/node-serialport #638: Calling `close()` removes all event listeners (including "error")

bminer, 2015-12-01:
When a SerialPort is closed by calling `close()`, all event listeners are removed (as per the documentation).  This behavior might be undesired.

If one calls `write` on a closed serial port connection without passing a callback to the `write` method, an "error" event is emitted by default.  So, calling `port.write(buf);` on a closed serial port can cause an "error" event to be emitted without an "error" event handler (even if the user created an "error" event handler before the SerialPort was closed).  In this case, the default Node behavior for this is to throw an exception, which could crash the program.

rwaldron, 2016-02-17:
Just hit this as well. Even if I have registered an error event handler, there are no events registered when this: https://github.com/voodootikigod/node-serialport/blob/master/serialport.js#L293 is reached—which means that code is pointless.

reconbot, 2016-03-28:
This bananas

rwaldron, 2016-03-28:
(╯‵Д′)╯彡┻━┻

reconbot, 2016-03-28:
Tracking in #702

closed 2016-04-25 by reconbot

reconbot, 2016-04-25:
fixed in 3.0.0

PR #764 (2016-04-25): Stop removing all event listeners on close and disconnect
Should address some of #702 and closes #638 and helps https://github.com/rwaldron/johnny-five/issues/1037