serialport/node-serialport #641: Closing and reopening the port.

mon1020, 2015-12-05:
Hi,

I am using the latest version of the module.
I am facing an issue below:
In my node program, if I disconnect the serial port, it tries to open it unless the serial port is found. Once found, it opens the device successfully. But give me error in writing to the serial port.
"Error: Error Input/output error calling write(...)"

Also, if I close my node program and run again, it works fine with the same connection.

Please suggest.

Thanks in advance!

Regards,
Monica

mon1020, 2015-12-07:
Hi  @voodootikigod,

Could you please help me resolve the issue..
I shall appreciate your quick reply.

Thanks in anticipation.

Regards,
Monica

rwaldron, 2016-02-17:
This is a duplicate of https://github.com/voodootikigod/node-serialport/issues/638

closed 2016-03-28 by reconbot