serialport/node-serialport #643: Parsers Readline Exemptions

ricxsar, 2015-12-08:
Hi! Is there any functionality on parser readline which will return data when a special character is received?

I have an application which uses ```readline('\r\n')``` as a delimeter. If the serial receive a special character ```>```, it should return the data.

Below is how the serialport is instantiated.
```
var serialport = require("serialport");
var SerialPort = serialport.SerialPort;

var serial = new SerialPort('/dev/ttyO4', {
	baudrate: 9600,
  parser: serialport.parsers.readline('\r\n'),
  flowControl: true
});

```

FF-Micheln, 2016-01-04:
You can check if this works for you. By setting the Parser you can add an additional second Parameter, called encoding, test if "ascii" or something else do the magic. Basicly the encoding is used in the method toString as parameter on the Buffer.

General: https://nodejs.org/api/buffer.html#buffer_buf_tostring_encoding_start_end
readLine parser: https://github.com/voodootikigod/node-serialport/blob/master/parsers.js#L11

FF-Micheln, 2016-01-04:
You can store the data in an Variable and output it when the Char ">" is in the string.

```
var buff = '';
serial.on('data', function (msg) {
	buff += msg;
	if (buff.indexOf('>') !== -1) { //check if an ">" is in the buff Variable
		console.log(buff); //output the Buffer
		buff = ''; //clean the Buffer
	}
});
```

reconbot, 2016-03-30:
It sounds like you want to use the `readline` parser but with `>` instead of the `\r\n`.

closed 2016-03-30 by reconbot

leesohaeng, 2016-08-30:
I got a same problem.
but i think leadline in serial port module not support instead of the \r\n.