serialport/node-serialport #648: Auto Reconnenction on Serialpost while plug/unplug device

vishvas29, 2015-12-23:
Hello,

I am using GSM modem with RPi using USB serailport. When Cable is unplugged then plugged again connection can not be established again. so, Is it possible to reconnect modem automatically? 

Thank you.

sboudouk, 2016-03-24:
Same issues here.

reconbot, 2016-03-27:
There is no auto reconnect, but we currently have an issue with not being able to close and open ports under all circumstances. See #670

reconbot, 2016-04-06:
@vishvas29 @sboudouk We fixed a ton of bugs in our latest beta specifically ones addressing this issue, I think. It will now properly open and close serialports on osx and linux. I don't know what platform you're on.

https://github.com/voodootikigod/node-serialport/issues/733 Please try it out `serialport@2.0.7-beta3` and report back!

PS

The `close()` will remove all event handlers. that's a bad design and will change in the next major serialport version and will very much make reconnecting easier. We've documented this behavior so it's a lot more clear, and we'll have it changed in `serialport@3.0.0` probably soon after `2.07` lands.

reconbot, 2016-05-23:
I've added a "auto reconnect" feature to our backlog #746

I'm going to close this issue for now as I think you're in good shape for the closing and opening ports. Reopen or open a new issue if you still have issues!

closed 2016-05-23 by reconbot