serialport/node-serialport #649: failed to install node-serialport on Raspberry Pi2

ericdai, 2015-12-24:
I follow the document to install serialport module on Raspberry Pi2 ,but failed.
pi@raspberrypi:~ $ sudo npm install serialport
/
> serialport@2.0.5 install /home/pi/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/roo                                                            t/.node-gyp/0.12.6"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_m                                                            odules/serialport/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: read ECONNRESET
gyp ERR! stack     at exports._errnoException (util.js:746:11)
gyp ERR! stack     at TCP.onread (net.js:559:26)
gyp ERR! System Linux 4.1.13-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_mod                                                            ules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/home                                                            /pi/node_modules/serialport/build/Release/node-v14-linux-arm/serialport.node" "-                                                            -module_name=serialport" "--module_path=/home/pi/node_modules/serialport/build/R                                                            elease/node-v14-linux-arm"
gyp ERR! cwd /home/pi/node_modules/serialport
gyp ERR! node -v v0.12.6
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local                                                            /lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback                                                            -to-build --module=/home/pi/node_modules/serialport/build/Release/node-v14-linux                                                            -arm/serialport.node --module_name=serialport --module_path=/home/pi/node_module                                                            s/serialport/build/Release/node-v14-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/s                                                            erialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:1015:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_proces                                                            s.js:1087:5)
node-pre-gyp ERR! System Linux 4.1.13-v7+
node-pre-gyp ERR! command "node" "/home/pi/node_modules/serialport/node_modules/                                                            .bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/serialport
node-pre-gyp ERR! node -v v0.12.6
node-pre-gyp ERR! node-pre-gyp -v v0.6.13
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modu                                                            les/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/pi/nod                                                            e_modules/serialport/build/Release/node-v14-linux-arm/serialport.node --module_n                                                            ame=serialport --module_path=/home/pi/node_modules/serialport/build/Release/node                                                            -v14-linux-arm' (1)
npm ERR! Linux 4.1.13-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport"
npm ERR! node v0.12.6
npm ERR! npm  v2.11.2
npm ERR! code ELIFECYCLE

npm ERR! serialport@2.0.5 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@2.0.5 install script 'node-pre-gyp install --f                                                            allback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
pi@raspberrypi:~ $

how to process it ?

cheton, 2015-12-25:
I also encountered similar installation issue with Node.js v4.2 and serialport-2.0.5 since yesterday.

```bash
npm ERR! Linux 2.6.32-71.el6.x86_64
npm ERR! argv "/home/cheton/.nvm/versions/node/v4.2.2/bin/node" "/home/cheton/.nvm/versions/node/v4.2.2/bin/npm" "install"
npm ERR! node v4.2.2
npm ERR! npm  v2.14.7
npm ERR! code ELIFECYCLE

npm ERR! serialport@2.0.5 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@2.0.5 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/cheton/github/cnc.js/npm-debug.log
```

fivdi, 2015-12-27:
@ericdai does it work if 
```
sudo npm install --unsafe-perm serialport
```
is used rather than
```
sudo npm install serialport
```
Also, why is it necessary to use ```sudo```?

cheton, 2015-12-30:
I eventually fixed my installation problems after upgrading g++/gcc version to 4.8.

fivdi, 2015-12-30:
@cheton g++ 4.8 or higher is required for compiling addons with Node.js v4 or higher. @ericdai is using v0.12.6 and is having a different issue.

TascSystems, 2016-01-06:
I'm having a similar issue on a BeagleBone Black variant.

Logfile is here: https://www.dropbox.com/s/dihkjm43iy5r5kj/npm-debug.log?dl=0

Should I downgrade to node v0.10.x to resolve this issue? Is there a compatibility matrix that I can use to determine what version of node-serialport I should use with a version of node?

roccomuso, 2016-01-09:
Might be related to this https://github.com/voodootikigod/node-serialport/issues/658 ?

heymind, 2016-02-16:
Maybe you can upgrade your nodejs to 4.2 and npm to 3.+.Then do not use 'sudo'.
I had similar problems when I install it on my pi2(run dietpi).

jacobrosenthal, 2016-02-22:
As discussed above:
`npm install serialport` installs the lastest serialport, 2.x.x
However 2.x.x in order to add node4 support now requires gcc 4.8.x
The vast majority of raspberry pi users are on debian 'wheezy' with gcc 4.7.x
Debian 'jessie' added gcc 4.8.x support
So you either need to upgrade to jessie/upgrade to gcc 4.8 to use the latest serialport
or you need to install the 1.x version of serialport with `npm i serialport@1.7.4`

closed 2016-02-22 by jacobrosenthal

howardkitto, 2016-02-22:
Thanks for getting back to me so quickly but I’m not sure that’s the answer

Looks like I’m on

Jessie

and 

GCC 4.9

I’ll try 1.7.4

pi@raspberrypi:~/Documents/NodeProjects/webservice $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/Documents/NodeProjects/webservice $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Raspbian 4.9.2-10)

fivdi, 2016-02-22:
@howardkitto from issue https://github.com/voodootikigod/node-serialport/issues/673 it can be seen that `npm install serialport` results in the following error:

```
error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
```

see https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-Raspberry-Pi for further details related to this error.

howardkitto, 2016-03-02:
Hi 

I fixed my UTF8 problem but now I'm getting this error

Can you help?

> serialport@2.0.6 install /home/pi/Documents/NodeProjects/ledClient/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp ERR! configure error 
gyp ERR! stack Error: ENOSPC, mkdir '/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build'
gyp ERR! System Linux 4.1.17+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "configure" "--fallback-to-build" "--module=/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build/Release/node-v11-linux-arm"
gyp ERR! cwd /home/pi/Documents/NodeProjects/ledClient/node_modules/serialport
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp configure --fallback-to-build --module=/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build/Release/node-v11-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:755:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:822:5)
node-pre-gyp ERR! System Linux 4.1.17+
node-pre-gyp ERR! command "node" "/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/Documents/NodeProjects/ledClient/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.29
node-pre-gyp ERR! node-pre-gyp -v v0.6.18
node-pre-gyp ERR! not ok 
Failed to execute 'node-gyp configure --fallback-to-build --module=/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build/Release/node-v11-linux-arm' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
gyp ERR! configure error 
gyp ERR! stack Error: ENOSPC, mkdir '/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build'
gyp ERR! System Linux 4.1.17+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "configure" "--fallback-to-build" "--module=/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/pi/Documents/NodeProjects/ledClient/node_modules/serialport/build/Release/node-v11-linux-arm"
gyp ERR! cwd /home/pi/Documents/NodeProjects/ledClient/node_modules/serialport
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm ERR! serialport@2.0.6 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@2.0.6 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.17+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/Documents/NodeProjects/ledClient
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/Documents/NodeProjects/ledClient/npm-debug.log
npm ERR! not ok code 0

fivdi, 2016-03-02:
There's an ENOSPC error. Is the SD card full?

howardkitto, 2016-03-03:
Right, I've fixed the diskspace issue now I'm getting this, any ideas?

npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! serialport@2.0.6 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 127
npm ERR! 
npm ERR! Failed at the serialport@2.0.6 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.18+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/Projects/ledClient
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/Projects/ledClient/npm-debug.log
npm ERR! not ok code 0

devrobot01, 2016-05-31:
wget http://node-arm.herokuapp.com/node_latest_armhf.deb
dpkg -i node_latest_armhf.deb
npm install -g node-gyp
sudo npm install --unsafe-perm serialport

fixed my problem