serialport/node-serialport #650: Command failed error on listing ports (udevadm is missing)

dbaba, 2015-12-27:
Listing ports on Intel Edison Yocto Linux always fails.

Tested Version: 2.0.5/2.0.6

```
var serialp = require("serialport");
serialp.list(function (err, ports) { console.log(err,ports); });
```
This shows the following error message.
```
{ [Error: Command failed: /bin/sh: /sbin/udevadm: No such file or directory
/bin/sh: /sbin/udevadm: No such file or directory
] killed: false, code: 127, signal: null } [ undefined ]
```
However, udevadm does exist there:
```
$ which udevadm
/bin/udevadm
```

jacobrosenthal, 2015-12-27:
Looks like we expect it at /sbin/udevadm
https://github.com/voodootikigod/node-serialport/blob/master/serialport.js#L586

Try symlinking
ln -s /bin/udevadm /sbin/udevadm

Might need sudo

dbaba, 2015-12-27:
Thanks. Prior to reading your comment, I created a patch. Is it useful for the project?

https://github.com/dbaba/node-serialport/commit/eac192af47f17debfed731d91acdd70c81079e5e

reconbot, 2016-03-27:
I'd rather do it differently but Yocto should be supported without hacking. Move this to the backlog. I wonder if we can rely on `udevadm` being in the path.

reconbot, 2016-03-31:
Fixed in master, will be in the next beta =) Please it out and let us know if it works for you!

commit 3ef00549 (2016-03-30): Move listUnix to its own module, move stuff out of the factory

 - refactor a little to fix bugs but only in listUnix 
 - test listUnix
 - remove the path behind `udevadm` as it’s not always in the same place but it’s usually in the path
 - Tests everything but `pnpId` and assumes a lot about `udevadm`

Fixes #650 and #634
commit 00d5f078 (2016-03-30): Move listUnix to its own module, move stuff out of the factory

 - refactor a little to fix bugs but only in listUnix 
 - test listUnix
 - remove the path behind `udevadm` as it’s not always in the same place but it’s usually in the path
 - Tests everything but `pnpId` and assumes a lot about `udevadm`

Fixes #650 and #634
commit a396d3da (2016-03-30): Move listUnix to its own module, move stuff out of the factory

 - refactor a little to fix bugs but only in listUnix 
 - test listUnix
 - remove the path behind `udevadm` as it’s not always in the same place but it’s usually in the path
 - Tests everything but `pnpId` and assumes a lot about `udevadm`

Fixes #650 and #634
commit 2dc28beb (2016-03-30): Move listUnix to its own module, move stuff out of the factory

 - refactor a little to fix bugs but only in listUnix 
 - test listUnix
 - remove the path behind `udevadm` as it’s not always in the same place but it’s usually in the path
 - Tests everything but `pnpId` assumes a lot about `udevadm`

Fixes #650 and #634
closed 2016-03-31 by reconbot