serialport/node-serialport #651: Add more tty* path support for listLinux()

dbaba, 2015-12-27:
I've found that `listLinux()` function discards paths under `/dev/` other than `ttyS*`, `ttyACM*`, and `ttyUSB*`.

https://github.com/voodootikigod/node-serialport/blob/master/serialport.js#L579

Intel Edison Yocto, however, offers some serial ports which don't match the above filtering rules, e.g. `/dev/ttyMFD0`.

I think the line can be replaced with the following.
```javascript
if (!files[i].match(/tty[a-zA-Z]/) || !fs.statSync(path.join(dirName,files[i])).isCharacterDevice()){
```
How about that?

dbaba, 2015-12-30:
@voodootikigod I found the similar issue #610 
Should I re-open the issue with concrete port names and close this?
I can list the all required file names.

reconbot, 2016-03-27:
@dbaba a full list would be great. I'm not up to speed on this issue but it seems like something easy to tackle. Yocto is something I'd like to support better. We should also probably evaluate using `fs.statSync` =(

reconbot, 2016-03-31:
This one is fixed too, if you have more port names let us know!

closed 2016-03-31 by reconbot