serialport/node-serialport #652: Wrong binary hostName in package.json

uwburn, 2015-12-29:
I have been wondering for some time why i always had to rebuild everything even if i am working on a platform with published bins (win64 + node46). I had a few time to spare this morning and i found a typo in the package.json: binary.host is "https://github.com/voodootikigod/node-serialport/releases/download/2.0.5", but the correct url is "https://github.com/voodootikigod/node-serialport/releases/download/v2.0.5", there's a missing v before the version number.

jacobrosenthal, 2015-12-29:
Change the release names rather than the package name. Try now.

uwburn, 2015-12-30:
Works perfect now! I hope this will help win users without the full toolchain installed.

PS
Thanks for the great lib!

closed 2015-12-30 by uwburn