serialport/node-serialport #657: Install failure on Debian-based/Arm system

TascSystems, 2016-01-06:
I installed node using the following instructions: http://www.armhf.com/node-js-for-the-beaglebone-black/

I've tried installing serialport on three versions of node including v0.10.5, v0.10.41 and v0.12.x (latest), and consistently have the following installation error.

I've been able to npm install every other module I use - but serialport fails.

Wasn't sure if this was the same issue that others with similar reports are seeing, so I created a separate issue - sorry if this is a duplication.

```
root@debian-armhf:/home/debian/peakgps# npm install serialport --save
npm WARN deprecated This version of npm lacks support for important features,
npm WARN deprecated such as scoped packages, offered by the primary npm
npm WARN deprecated registry. Consider upgrading to at least npm@2, if not the
npm WARN deprecated latest stable version. To upgrade to npm@2, run:
npm WARN deprecated
npm WARN deprecated   npm -g install npm@latest-2
npm WARN deprecated
npm WARN deprecated To upgrade to the latest stable version, run:
npm WARN deprecated
npm WARN deprecated   npm -g install npm@latest
npm WARN deprecated
npm WARN deprecated (Depending on how Node.js was installed on your system, you
npm WARN deprecated may need to prefix the preceding commands with `sudo`, or if
npm WARN deprecated on Windows, run them from an Administrator prompt.)
npm WARN deprecated
npm WARN deprecated If you're running the version of npm bundled with
npm WARN deprecated Node.js 0.10 LTS, be aware that the next version of 0.10 LTS
npm WARN deprecated will be bundled with a version of npm@2, which has some small
npm WARN deprecated backwards-incompatible changes made to `npm run-script` and
npm WARN deprecated semver behavior.
npm WARN package.json peakgps@1.0.0 No repository field.
npm WARN package.json peakgps@1.0.0 No README data

> serialport@2.0.6 install /home/debian/peakgps/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.10.41"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/debian/peakgps/node_modules/serialport/.node-gyp"
make: Entering directory `/home/debian/peakgps/node_modules/serialport/build'
make: *** No rule to make target `../.node-gyp/0.10.41/common.gypi', needed by `Makefile'.  Stop.
make: Leaving directory `/home/debian/peakgps/node_modules/serialport/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Linux 3.14.1_001-peak
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/debian/peakgps/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/debian/peakgps/node_modules/serialport/build/Release/node-v11-linux-arm"
gyp ERR! cwd /home/debian/peakgps/node_modules/serialport
gyp ERR! node -v v0.10.41
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/debian/peakgps/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node --module_name=serialport --module_path=/home/debian/peakgps/node_modules/serialport/build/Release/node-v11-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/debian/peakgps/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:766:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:833:5)
node-pre-gyp ERR! System Linux 3.14.1_001-peak
node-pre-gyp ERR! command "node" "/home/debian/peakgps/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/debian/peakgps/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.41
node-pre-gyp ERR! node-pre-gyp -v v0.6.18
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/debian/peakgps/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node --module_name=serialport --module_path=/home/debian/peakgps/node_modules/serialport/build/Release/node-v11-linux-arm' (1)

npm ERR! serialport@2.0.6 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@2.0.6 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.14.1_001-peak
npm ERR! command "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport" "--save"
npm ERR! cwd /home/debian/peakgps
npm ERR! node -v v0.10.41
npm ERR! npm -v 1.4.29
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0
```

TascSystems, 2016-01-07:
Looks like I was able to install this using the following guidelines provided by jywarren (2nd comment) here: https://github.com/mapbox/node-pre-gyp/issues/165

He says: 
```
Those errors look related to permissions only. Are you perhaps running as root? If so can you run as a normal user? I'm aware that npm does not work correctly when run as root unless you also pass --unsafe-perm npm/npm#3497.

That said if this is not the issue happy to try to help more. Basically node-pre-gyp and node-gyp both need to be callable as command line tools. So if permissions are off they often break, but the underlying problem can only be fixed by fixing permissions.
```

JoseAlvaradoRuiz, 2016-01-07:
How exactly you were able to install serialport? Did you use a normal user or you used the --unsafe-perm npm/nom#3497 lines in the instalation?

TascSystems, 2016-01-07:
I installed node-gyp and node-pre-gyp first, then installed serialport with the --unsafe-perm option. I also installed using sudo. 

In retrospect, I'm not sure if I needed to install node-gyp or node-pre-gyp, but the combination allowed it to install. I think the key was the --unsafe-perm option.

fernfrost, 2016-02-05:
Had the same problem. Turned out i had to install nodejs-legacy

PaulBernier, 2016-02-19:
Ffor me the solution was installing nodejs-legacy +  --unsafe-perm option for npm. Thanks guys!

reconbot, 2016-04-06:
Glad it worked out!

closed 2016-04-06 by reconbot