serialport/node-serialport #659: My result are broken

abraxasband, 2016-01-10:
SO: Windows10
Hardware: Arduino Mega256
Server: NodeJS

Hi All
My problem is that my data appears divided. 

My arduino code is simple

Serial.print("Hola");
delay(1000);

My code in Node 
[image]

But my data ought to be only "Hola" and the result is this

[image]

I receipt data, but for some reason, they are cut.

any ideas?

Thanks!

jacobrosenthal, 2016-01-11:
Serial data is delivered at its convenience. You need collect the results into a string or a buffer and sort through.  Or you can use the inbuilt parser to parse for a delimeter of your choice. You have commented out a readline parser which will wait to deliver your string until it finds a newline.

closed 2016-01-11 by jacobrosenthal

abraxasband, 2016-01-11:
Thanks Jacob Rosenthal!