serialport/node-serialport #660: Abort trap: 6

claudio-destro, 2016-01-23:
Hi guys

Chatted with @reconbot about the following error I got in my code:

Assertion failed: (loop->watchers[w->fd] == w), function uv__io_stop, file ../deps/uv/src/unix/core.c, line 864.
Abort trap: 6

it seems to be random... I got it in my library [flashmagic.js](https://github.com/claudio-destro/flashmagic.js) branch develop.

please run cli.ts with subcommand "ping" but I think you will need an NXP LPC hardware...

```bash
# ./bin/flashmagic.js -P <your_port> ping
```

I get the error after attempting to close and reopen a given serial port with different baud rate (see last lines in src/Handshake.ts): 9600 -> 115200.

---

Attempting to reproduce the issue I wrote the following code in plain javascript: **it fails fast** sometimes with [Error: Serialport not open.] (uncomment on('data')), sometimes with [Error: Cannot open /dev/tty.usbserial-FTBTBAAW].

It seems like the close/open functions do not work as expected.

```javascript
var com = require('serialport');

var path = process.argv[2]

var serialport = openCom(path, 9600);
var count = 1;

(function loop() {
  // serialport.on('data', function(data) {
  //   console.log(data);
  // });
  serialport.open(function(error) {
    if (error) { console.error(error); throw error;}
    serialport.write('?', function(error) {
      if (error) { console.error(error); throw error;}
      serialport.drain(function(error) {
        if (error) { console.error(error); throw error;}
        serialport.close(function(error) {
          if (error) { console.error(error); throw error;}
          console.log('closed ' + count++);
          serialport = openCom(path, 115200);
          process.nextTick(loop);
        }); // close
      }); // drain
    }); // write
  });
})();

function openCom(path, baud) {
  return new com.SerialPort(path, {
      baudRate: baud,
      stopBits: 1,
      parity: 'none',
      parser: com.parsers.readline('\r\n')
    }, false); // open later
}

/*
$ uname -a
Darwin iMac-di-Claudio.lan 15.2.0 Darwin Kernel Version 15.2.0: Fri Nov 13 19:56:56 PST 2015; root:xnu-3248.20.55~2/RELEASE_X86_64 x86_64
$ node -v
v5.4.0
*/
```

claudio-destro, 2016-01-24:
See this
https://github.com/nodejs/node/issues/3604

reconbot, 2016-03-27:
Can you confirm your platform? I think this is related to #670

claudio-destro, 2016-03-27:
Confirmed. I think it is related to open/close port too: actually I have to use only one port at a time. 

Inviato da iPad

reconbot, 2016-03-27:
I meant, is this Linux on an embedded device, and if so, which one?

claudio-destro, 2016-03-27:
Nope

As written in the code, it is OS X with node 5.4

Claudio

reconbot, 2016-04-06:
@claudio-destro We fixed a ton of bugs in our latest beta specifically ones addressing this issue on osx.

https://github.com/voodootikigod/node-serialport/issues/733 Please try it out `serialport@2.0.7-beta3` and report back!

claudio-destro, 2016-04-06:
Ok I will try it out!

Inviato da iPhone

reconbot, 2016-04-11:
Did it work out? We've released `serialport@2.1.0`.

claudio-destro, 2016-04-11:
I didn't have enough time... Sorry

I did a little test yesterday night and seemed working. I will try the new release. 

Claudio

Inviato da iPhone

reconbot, 2016-05-16:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to reopen.

closed 2016-05-16 by reconbot

lucdetellis, 2016-11-09:
I was getting the "Abort trap: 6" crash. After some experimenting I found that I was sending writes one immediately after another, which was apparently too fast for serialport to handle.

Here's the setTimeout I had to add:
```
serial.write(output + '\n', function(err) {
  if (err) return log.error('serialport', err);
  setTimeout(function() {
    if (cue.length === 0) {
      canSend = true;
    } else {
      write(cue.shift());
    }
  }, 0);
});
```

reconbot, 2016-11-09:
Can you show me the crashing code? And can you tell me what version you're
running.

---
Francis Gulotta
wizard@roborooter.com

lucdetellis, 2016-11-09:
@reconbot Sorry I deleted my original comment, I found a better solution than using a setTimeout.

Here's my original code if you're interested:
```
var write = function(output) {
  if (serial.isOpen) {
    serial.write(output + '\n', function(err) {
      if (err) return log.error('serialport', err);
      if (cue.length === 0) {
        canSend = true;
      } else {
        write(cue.shift());
      }
    });
  }
};

var send = function(output) {
  if (canSend === true) {
    canSend = false;
    write(output);
  } else {
    cue.push(output);
  }
};

send('a lot of things at once');
...
send('more things');
```

I fixed the "Abort trap: 6" error by adding serial.drain() before each write:
```
var write = function(output) {
  serial.write(output + '\n', function(err) {
    if (err) return log.error('serialport', err);
    serial.drain(function(err) {
      if (err) return log.error('serialport', err);
      if (cue.length === 0) {
        canSend = true;
      } else {
        write(cue.shift());
      }
    });
  });
};
```

reconbot, 2016-11-09:
There isn't an upper limit on how many writes you can do in sequence, so
I'd like to get to the bottom of this.

---
Francis Gulotta
wizard@roborooter.com